FreeCAD/FreeCAD #5771: Array and dictionary lookup does not work in expressions

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3391

* **Reporter:** eivindkvedalen
* **Date submitted:** 3/24/2018
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

See discussion on https://forum.freecadweb.org/viewtopic.php?f=3&t=26876#p214826

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by ezzieyguywuf 2019-12-03 04:29
Is this a bug or a feature request?

As [discussed in this post](https://forum.freecadweb.org/viewtopic.php?f=10&t=8802&start=210#p215328), it sounds like there is a true security bug related to this, but I don't fully understand how to reproduce the issue that results in the security vulnerability.

Can someone provide a step-by-step to reproduce the [fix posted by triplus here](https://forum.freecadweb.org/viewtopic.php?f=3&t=26876#p214826), and describe how this leads to the security issue mentioned by elvind?

---

### Comment by Kunda1 2020-02-23 19:53
realthunder pardon, just wondering if your pending p:FreeCAD:3062: addresses this ?

---

### Comment by realthunder 2020-02-23 21:47
Yes, the PR addressed this. And it shall have completion support for that. And no, there won't be any security concern as mentioned in the referenced post, because I have already modified Expression/ObjectIdentifier to not use the Python interpreter.

---

### Comment by Kunda1 2020-02-23 23:03
realthunder wow, that's awesome news! There are other expression/spreadsheet issues here on the tracker. I wonder, and I want to be mindful of your time, if we can go through them and see if your PR also addresses them ?

Edit: If the PR does address and solve them, would it be possible to use the MantisBT wildcard triggers in the Git commit to remotely close said issues ?
See https://www.freecadweb.org/wiki/Tracker#GitHub_and_MantisBT

---

### Comment by realthunder 2020-02-24 01:53
I'll go through them when I get time. But some of them may be fixed already in upstream. I can't modify the commit message there. Would it be enough for me to just refer to the relevant commit in the issue comment?

---

### Comment by realthunder 2020-02-27 05:56
Extended indexing support is added with [this](https://github.com/FreeCAD/FreeCAD/commit/a1417c5ffa52ea1c55bda08f3534a098dae23e07) commit in upstream.

maxwxyz, 2024-12-07:
@luzpaz you have linked a PR some time ago, is this still relevant?

luzpaz, 2025-01-25:
No. It stalled. I'll unlink.

closed 2025-06-08 by github-actions[bot]