FreeCAD/FreeCAD #5775: Hiding / Showing features hides entire body

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3398

* **Reporter:** pager48
* **Date submitted:** 3/26/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Cannot hide and show features repeatedly, what happens is entire body is hidden.

# Steps to reproduce

Make some 3D solid
Make a fillet on some edge
Select fillet from model panel
Hit space to hide it

Results: Entire body is hidden
What should have happened is the filled feature becoming hidden.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by pager48 2018-03-26 23:03
This works with many other feature types as well, including chamfer.

---

### Comment by pager48 2018-03-27 03:34
Freecad revision:

OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13452 (Git) AppImage
Build type: None
Branch: master
Hash: 34633c144de9133c1f9aeb7da783f369cae9bfaf
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)

---

### Comment by Kunda1 2018-03-29 10:55
chrisb care to respond to this ticket? Thanks

---

### Comment by pager48 2018-03-30 08:27
It should work regardless of how many times visibility is toggled.

---

### Comment by DeepSOIC 2018-04-04 22:18
This is by design. When a feature is shown, the whole state of the solid after applying the feature is shown. Hide/Show is not going to switch features on and off in the middle of modeling history.

What can be done, is when you hide Fillet, a previous feature is made visible automatically. I would regard this as a valid feature request, but not a bug.

---

### Comment by DeepSOIC 2018-04-04 22:20
pager48, please don't pollute the tracker, and ask on forum first!

---

### Comment by Kunda1 2018-04-15 17:10
Changing to feature request per DeepSOIC's recommendations

---

### Comment by Kunda1 2020-10-15 13:59
This feature may be already implemented in realthunder's LinkStage3 branch:
Right click on a feature in the tree -> suppress

adrianinsaval, 2022-02-10:
my understanding is that this should be closed as everything is working as intended. It's expected that hiding the tip will hide the entire shape. Supressing is a different feature and IMO should not be triggered by space key.

sloworks, 2023-02-15:
LinkStage3 is missing still small enhancement: ability to use spreadsheet to switch suppress/unsuppress state

closed 2023-02-15 by luzpaz