FreeCAD/FreeCAD #5776: Implement dimension tool

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3399

* **Reporter:** pager48
* **Date submitted:** 3/26/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Currently freecad requires a user to select appropriate constraint tool to constrain an object. This is because no dimension tool exists, such a tool would automatically detect object type and apply appropriate constraint rather than causing user to select appropriate constraint tool every time they want to constrain an object. The dimension tool should remain enabled to permit user to dimension as many items as desired thus not having to toggle the tool for every item they need to constrain.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

adrianinsaval, 2022-02-11:
duplicate of #5582 and tackled by #5407 can be closed IMO

closed 2022-02-12 by chennes

luzpaz, 2022-02-12:
#5407 hasn't passed review, I think the last commit in that PR can be used auto-close this ticket:   
`Closes #5776`

adrianinsaval, 2022-02-12:
I suggested closing for being duplicate, not because it's merged