FreeCAD/FreeCAD #5779: Provide Attachment mode perpendicular to an existing plane

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3407

* **Reporter:** chrisb
* **Date submitted:** 3/28/2018
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Attachment should support attachin a plane perpendicular to another plane.
[See forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=27698)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

Roy-043, 2024-06-13:
#14372 introduces the "XY parallel to plane" mode. With angular offsets that mode can probably be used to align perpendicularly as well.

Roy-043, 2024-06-13:
@chrisb-github Should this stay open?

luzpaz, 2024-08-02:
@chrisb-github bump

maxwxyz, 2024-09-04:
I'm closing this then fo rno feedback and it is possible as described above.

closed 2024-09-04 by maxwxyz

luzpaz, 2024-09-04:
JFYI @chrisb-github