FreeCAD/FreeCAD #5781: SVG imports from inkscape lead to wrong dimensions

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3420

* **Reporter:** herrdeh
* **Date submitted:** 4/3/2018
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** assigned
* **Tags:** import, Inkscape, SVG

# Original report text

With the change from 0.91 to 0.92 inkscape devs changed the conversion factor of userunit (uu) and mm. Pre-0.92 it was 1uu=1 inch/90 and starting from 0.92 it is 1 uu=1inch/96. This was necessary to meet the actual recommendations/rules of svg-standard and css. See [this article](http://wiki.inkscape.org/wiki/index.php/Units_In_Inkscape) in inkscapewiki for further reading. 

When a user creates a file in inkscape with absolute dimensions (mm) and saves it, then imports it into FC, bodies in Freecad come out with too large dimensions. For example - a 100mm square will be 106,66mm in FC now (100 x 96/90).

[In this discussion in the German forum](https://forum.freecadweb.org/viewtopic.php?f=13&t=27344) folks who can read German can follow the process of isotating the bug.

New forum thread opened in https://forum.freecadweb.org/viewtopic.php?f=8&t=37102

# Steps to reproduce

Import a SVG file from inkscape >=0.92 - check measures.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: all all**
* **Platform:** all
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by yorik 2018-04-03 15:38
Would it be possible to provide two very simle test files that would allow to check the results?

---

### Comment by herrdeh 2018-04-04 08:14
Here are inkscape test files. Unluckily, none of my FC versions are capable of SVG import at the moment, so I cannot provide FC test files. But importing both of these examples should lead to different squares, allthough both of them are 100x200mm in inkscape.

---

### Comment by yorik 2018-04-04 14:07
Thanks! I'll have a look at this after the release is out.

---

### Comment by Kunda1 2019-06-17 14:59
New forum discussion thread opened in https://forum.freecadweb.org/viewtopic.php?f=8&t=37102

---

### Comment by teo8976 2019-11-23 13:40
This has nothing to do specifically with Inkscape, and it has nothing to do specifically with mm.

Look at the attached SVG file. I wrote this SVG file by hand, it was not exported from Inkscape:

`<svg width="595.28pt" height="420.94pt"  viewBox="0 0 210 148" xmlns="http://www.w3.org/2000/svg">`

This defines a physical size in pt for the document (which is A4) and a corresponding viewbox such that 1 user unit should be 2.83465 pt which is 1 mm.

Hence the physical size of 1 unit in the document should be that.
That is defined by the document.

I have imported this and exported to dxf and the resulting width of the 190mm rectangles is about 185mm.

Then for some reason, if instead of exporting to dxf I print it to pdf (not export to pdf, print and select "print to pdf"), for some reason they're even much smaller.
But I don't know, maybe the "print" feature has no notion of a real physical size and just print everything to fit whatever paper size you choose.

---

### Comment by openBrain 2019-11-24 13:30
teo8976 : actually it has to do with Inkscape because for long time, FreeCAD relies on Inkscape SVG format. Inkscape doesn't export a viewbox so user unit can't be arithmetically determined. Issue was that Inkscape was using a strange 90 dpi so FC converts with this setting. Inkscape has recently moved to SVG standard 96 dpi and it makes emerge some problems. Things has been greatly improved and I'm now able to correctly import your file with latest 0.19 (despite I found a small bug that I'll report). Time for you to leave your 0.15 release and upgrade. ;)

---

### Comment by teo8976 2019-11-24 13:54
> Inkscape doesn't export a viewbox so user unit can't be arithmetically determined

Yeah but my file (which was not created by Inkscape) does have a viewbox AND document width and height on the root node, so the size of a user unit should be calculated from those without any fixed dpi involved. I haven't tried changing the width, height and viewbox though, to see whether the results differ (they should).

> Time for you to leave your 0.15 release and upgrade

I know, that's the version I got with apt-get. Time for [whoever maintains that] to release a more recent one for Ubuntu.

---

### Comment by openBrain 2019-11-24 14:20
teo8976 : the file you posted in this ticket raised the evidence that 'pt' unit has been forever not well defined in FC. ;) I submitted [this PR](https://github.com/FreeCAD/FreeCAD/pull/2733) to fix it. However as I explained above, this only solves part of your problems as I explained above.
Everything should be OK with 0.19 version once the PR is integrated. Thanks for your perseverance.

luzpaz, 2022-05-04:
@0penBrain anything still remaining with this ticket/issue?

0penBrain, 2022-05-04:
@luzpaz Not that I know of. ;)

luzpaz, 2022-05-04:
Fixed by @0penBrain in b809867.  Thanks!  
Closing.

closed 2022-05-04 by luzpaz