FreeCAD/FreeCAD #5782: Boolean operation in PDN fails depending on placement of one of the parts

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3425

* **Reporter:** nikisteen
* **Date submitted:** 4/9/2018
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #ready-for-migration, boolean, OCC 7.4, OCC Bug, OCC NotOnRoadmap, Part Design NEXT, upstream

# Original report text

When creating a boolean fuse between a cylinder and sphere (as shown in the attached project), the sphere disappear when its placement is adjusted to certain values, although the sphere still intersects the cylinder.  

# Additional information

Refer to [forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=27958) 

Freecad version information:
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13488 (Git)
Build type: Release
Branch: master
Hash: 2891173923bcecaace64e0544f5773facbdef436
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/SouthAfrica (en_ZA)

It was also confirmed to occur on the newest stable release (0.17.13509) - see forum thread.

# Steps to reproduce

In the attached project, adjust the Y-axis placement of the sphere to 8mm.  The sphere will be offset as expected.  Now adjust the Y-axis placement to 9mm and the sphere disappears. 

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: Windows 10**
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2018-06-07 16:57
wmayer 
DeepSOIC thinks this is a OCC bug. Please weigh in. Thanks!

---

### Comment by wmayer 2018-09-26 09:35
Seems to be related to a different behaviour between OCC 7.2 and older versions. With 7.2 I can confirm the behaviour but with 7.0 it works as expected.

---

### Comment by Kunda1 2018-09-26 13:24
wmayer what about OCC 7.3?

---

### Comment by wmayer 2019-02-20 15:44
There is a different behaviour in 7.2. For some reason the fusion of the sphere with the fused cylinder and box creates a compound with two solids.

```
cyl=App.ActiveDocument.Cylinder.Shape
sphere=App.ActiveDocument.Body002.Shape
cube=App.ActiveDocument.Body001.Shape
fuse=cyl.fuse(cube)
fuse.Solids
# shows one solid
fuse=fuse.fuse(sphere)
fuse.Solids
# shows two solids => error
```

If the y value of the placement of the sphere is <= 8 then the second fusion correctly creates a single solid.

---

### Comment by normandc 2019-02-24 15:45
Same issue with OCC 7.3.

OS: Ubuntu 18.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15972 (Git)
Build type: Release
Branch: master
Hash: 4e973324ae03c6db9188c08483232fa059c1c9ae
Python version: 3.6.7
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/Canada (fr_CA)

---

### Comment by Kunda1 2019-06-17 07:54
Added to "OCC Bugs in the Bugtracker" [thread](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264&p=156236#p156236)

---

### Comment by Kunda1 2022-01-15 13:17
user1234 writes in https://forum.freecadweb.org/viewtopic.php?p=534379#p534379
[quote=user1234]I think this bug is solved (and was not OCCT related). When i switch the visibility to Tools and then to Result back, the file is fine. Also when i delete the boolean operation and redo it, it works fine.
[/quote]

---

### Comment by Kunda1 2022-01-23 00:21
See responses:
https://forum.freecadweb.org/viewtopic.php?p=561983#p561983
https://forum.freecadweb.org/viewtopic.php?p=562493#p562493

luzpaz, 2022-07-13:
Per @adrianinsaval in the [forum thread](https://forum.freecadweb.org/viewtopic.php?p=562493#p562493):
> the boolean in the file [BoolTestWorks.zip](https://github.com/FreeCAD/FreeCAD/files/9105740/BoolTestWorks.zip) apparently doesn't work even if you recompute, redoing the boolean initially makes it work, also with y `placement=9mm`

>  but `y=8mm` makes it fail maybe even worse than before :? (with `refine=true`)   
> [BoolTestFail.zip](https://github.com/FreeCAD/FreeCAD/files/9105756/BoolTestFail.zip)

>  with `refine=false` I think it's more like the issue described
> [BoolTestFailNoRefine.zip](https://github.com/FreeCAD/FreeCAD/files/9105762/BoolTestFailNoRefine.zip)
> However the bug report says it should fail with `9mm` not `8mm`
 
```
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.26761 (Git)
Build type: Release
Python version: 3.8.12
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: Spanish/Paraguay (es_PY) 
```

grubertechnic, 2023-07-04:
That was probably an issue from an old version, that was nested in the file. When you change the boolean property Display from Result to Tools and then back, it works flawless. This is probably also no OCCT issue.

[2023-07-04 17-43.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/fa69e7c8-79a5-42ff-9e77-89593954ebda)

Also as i see, this was already tested in 2021 by me, see https://forum.freecad.org/viewtopic.php?p=534379&#p534379

@luzpaz and @adrianinsaval : probably a candidate to close?

closed 2023-08-28 by adrianinsaval