FreeCAD/FreeCAD #5783: Crash with STEP export (Upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3435

* **Reporter:** fhackenberger
* **Date submitted:** 4/19/2018
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #pending, #ready-for-migration, OCC 7.6, OCC Bug, STEP, upstream

# Original report text

Freecad crashes when exporting the (only) part in the attached project as a .step file.

OS: Ubuntu 16.04.4 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13515 (Git)
Build type: None
Branch: releases/FreeCAD-0-17
Hash: e17b340949b75a226cc7d89989b0aa238ccfc75f
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedKingdom (en_GB)

# Additional information

I exported the part as an IGES, re-imported it and then everything is fine.

$ freecad --log-file -
FreeCAD 0.17, Libs: 0.17R13515 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Mod pluginloader InitGui.py starting ...
pluginmanager config file /home/fhackenberger/.FreeCAD//Mod/plugins/pluginloaderconfig.yaml
pluginmanager userconfig file /usr/lib/freecad/Mod/plugins/myconfig.yaml
userconfigfile not available
IOError: [Errno 2] No such file or directory: '/usr/lib/freecad/Mod/plugins/myconfig.yaml'

pluginmanager userconfig file /usr/lib/freecad/Mod/plugins/myconfig.yaml
userconfigfile not available and cannot be created
IOError: [Errno 2] No such file or directory: '/usr/lib/freecad/Mod/plugins/myconfig.yaml'

Mod pluginloader InitGui.py done
gentoolbars ...
My Macros done
My Workbenches done
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x354b0) [0x7f190abe04b0]
#1  0x7f18e3ac65ff in TopoDSToStep_MakeStepFace::Init(TopoDS_Face const&, TopoDSToStep_Tool&, opencascade::handle<Transfer_FinderProcess> const&) from /usr/lib/x86_64-linux-gnu/libTKSTEP.so.7+0x1a7f
#2  0x7f18e3ab8573 in TopoDSToStep_Builder::Init(TopoDS_Shape const&, TopoDSToStep_Tool&, opencascade::handle<Transfer_FinderProcess> const&) from /usr/lib/x86_64-linux-gnu/libTKSTEP.so.7+0x503
#3  0x7f18e3ab8fe8 in TopoDSToStep_Builder::TopoDSToStep_Builder(TopoDS_Shape const&, TopoDSToStep_Tool&, opencascade::handle<Transfer_FinderProcess> const&) from /usr/lib/x86_64-linux-gnu/libTKSTEP.so.7+0x38
#4  /usr/lib/x86_64-linux-gnu/libTKSTEP.so.7(+0x260cfc) [0x7f18e3abecfc]
#5  0x7f18e3abf60f in TopoDSToStep_MakeManifoldSolidBrep::TopoDSToStep_MakeManifoldSolidBrep(TopoDS_Solid const&, opencascade::handle<Transfer_FinderProcess> const&) from /usr/lib/x86_64-linux-gnu/libTKSTEP.so.7+0x5f
#6  0x7f18e3adfe1c in STEPControl_ActorWrite::TransferShape(opencascade::handle<Transfer_Finder> const&, opencascade::handle<StepShape_ShapeDefinitionRepresentation> const&, opencascade::handle<Transfer_FinderProcess> const&, opencascade::handle<TopTools_HSequenceOfShape> const&, bool) from /usr/lib/x86_64-linux-gnu/libTKSTEP.so.7+0x1e7c
#7  0x7f18e3ae5057 in STEPControl_ActorWrite::Transfer(opencascade::handle<Transfer_Finder> const&, opencascade::handle<Transfer_FinderProcess> const&) from /usr/lib/x86_64-linux-gnu/libTKSTEP.so.7+0x2e7
#8  0x7f18e6406317 in Transfer_ActorOfFinderProcess::Transferring(opencascade::handle<Transfer_Finder> const&, opencascade::handle<Transfer_ProcessForFinder> const&) from /usr/lib/x86_64-linux-gnu/libTKXSBase.so.7+0x67
#9  0x7f18e6411696 in Transfer_ProcessForFinder::TransferProduct(opencascade::handle<Transfer_Finder> const&) from /usr/lib/x86_64-linux-gnu/libTKXSBase.so.7+0xc6
#10  0x7f18e641357c in Transfer_ProcessForFinder::Transferring(opencascade::handle<Transfer_Finder> const&) from /usr/lib/x86_64-linux-gnu/libTKXSBase.so.7+0x1cc
#11  0x7f18e6413da6 in Transfer_ProcessForFinder::Transfer(opencascade::handle<Transfer_Finder> const&) from /usr/lib/x86_64-linux-gnu/libTKXSBase.so.7+0x26
#12  /usr/lib/x86_64-linux-gnu/libTKXSBase.so.7(+0x177c53) [0x7f18e6497c53]
#13  0x7f18e6498758 in XSControl_Controller::TransferWriteShape(TopoDS_Shape const&, opencascade::handle<Transfer_FinderProcess> const&, opencascade::handle<Interface_InterfaceModel> const&, int) const from /usr/lib/x86_64-linux-gnu/libTKXSBase.so.7+0x78
[... 99 lines cut ...]
OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23237 (Git)
Build type: Debug
Branch: master
Hash: 71726a11d0844f263dad636fb98bb996a479e007
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

---

### Comment by Kunda1 2020-12-31 12:44
Tracking this via [OCC bugs in the bugtracker](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264&start=40) forum thread

---

### Comment by Kunda1 2022-01-15 13:21
user1234 writes in https://forum.freecadweb.org/viewtopic.php?p=534386#p534386
> When i open the file crash-step-export.fcstd and make a geometry check, the geometry is invalid. This is a common error of many exporters, i get that regularly from many manufacturers.
> 
> With the Defeaturing of the Part Workbench, i deleted the round surface, that was invalid and then the area is fine (with the bore, means no area is missing). Then the checks are good and exports and imports fine.

luzpaz, 2022-05-27:
@chennes @easyw Do you mind retesting to see if latest OCC (v7.6.2) has addressed this?

dpasukhi, 2024-09-25:
Hello, there is a OCCT bug for that issue?

dpasukhi, 2024-09-25:
Thank you. I will try to collect them for development plans (with current resources).
Additionally, you are free to create a tickets in GitHub.
_
Thank you for collecting all tickets in the single place!

dpasukhi, 2024-09-26:
Is it possible to share the project file as a .brep?
The related issue: https://github.com/Open-Cascade-SAS/OCCT/issues/80

The result of FreeCAD team investigation - file is not valid. But from our side it is necessary to avoid crash even on invalid files.

luzpaz, 2024-10-16:
> Is it possible to share the project file as a .brep?

Any volunteers ?

ElementW, 2026-08-24:
Reuploading (zip-repaired) document file here: [crash-step-export.fcstd.zip](https://github.com/user-attachments/files/31396174/crash-step-export.fcstd.zip)

Immediate crash on open with faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and OCC 8.1.0.dev1-3d097a0328
<details>
<summary>Stack trace</summary>

```
#0  0x00007fff93cc4847 in TopTools_ShapeSet::Read(TopoDS_Shape&, std::basic_istream<char, std::char_traits<char> >&, int) const () from /media/source/OCCT/build/release/usr/local/lib/libTKBRep.so.8.1
#1  0x00007fff93cc57c2 in TopTools_ShapeSet::Read(std::basic_istream<char, std::char_traits<char> >&, Message_ProgressRange const&) () from /media/source/OCCT/build/release/usr/local/lib/libTKBRep.so.8.1
#2  0x00007fff93cfa2fa in BRepTools::Read(TopoDS_Shape&, std::basic_istream<char, std::char_traits<char> >&, BRep_Builder const&, Message_ProgressRange const&) ()
   from /media/source/OCCT/build/release/usr/local/lib/libTKBRep.so.8.1
#3  0x00007fff9b7ce22d in Part::PropertyPartShape::loadFromStream (this=0x55555b6a8fd8, reader=...) at /media/source/FreeCAD/src/Mod/Part/App/PropertyTopoShape.cpp:701
#4  0x00007fff9b7cea29 in Part::PropertyPartShape::RestoreDocFile (this=0x55555b6a8fd8, reader=...) at /media/source/FreeCAD/src/Mod/Part/App/PropertyTopoShape.cpp:771
#5  0x00007ffff26225d4 in Base::XMLReader::readFiles (this=0x7fffffff9da0, zipstream=...) at /media/source/FreeCAD/src/Base/Reader.cpp:472
#6  0x00007ffff35a6f17 in App::Document::restore (this=0x5555584488c0, filename=0x55555b642590 "/tmp/fc/crash-step-export.fcstd", delaySignal=true, objNames=std::vector of length 0, capacity 0)
    at /media/source/FreeCAD/src/App/Document.cpp:2207
#7  0x00007ffff3926913 in App::Application::openDocumentPrivate (this=0x55555585bd30, FileName=0x5555596e9720 "/tmp/fc/crash-step-export.fcstd", propFileName=0x5555596e9720 "/tmp/fc/crash-step-export.fcstd", 
    label=0x55555842e520 "crash-step-export", isMainDoc=true, initFlags=..., objNames=...) at /media/source/FreeCAD/src/App/Application.cpp:1254
#8  0x00007ffff39251f2 in App::Application::openDocuments (this=0x55555585bd30, filenames=std::vector of length 1, capacity 1 = {...}, paths=0x0, labels=0x0, errs=0x0, initFlags=...)
    at /media/source/FreeCAD/src/App/Application.cpp:1033
#9  0x00007ffff392413b in App::Application::openDocument (this=0x55555585bd30, FileName=0x55555845b790 "/tmp/fc/crash-step-export.fcstd", initFlags=...) at /media/source/FreeCAD/src/App/Application.cpp:922
#10 0x00007ffff39aa2d6 in App::ApplicationPy::sOpenDocument (args=0x7fffeac3f940, kwd=0x0) at /media/source/FreeCAD/src/App/ApplicationPy.cpp:148
#11 0x00007ffff39aa177 in App::ApplicationPy::sOpen (self=0x7fffebb6fe20, args=0x7fffeac3f940, kwd=0x0) at /media/source/FreeCAD/src/App/ApplicationPy.cpp:117
#12 0x00007ffff11cf321 in cfunction_call (func=0x7fffebb88590, args=0x7fffeac3f940, kwargs=0x0) at Objects/methodobject.c:564
#13 0x00007ffff11ade71 in _PyObject_MakeTpCall (tstate=0x7ffff157a140 <_PyRuntime+315680>, callable=0x7fffebb88590, args=0x7fffffffaaf8, nargs=1, keywords=0x0) at Objects/call.c:242
#14 0x00007ffff11fd9bf in _PyEval_EvalFrameDefault (tstate=0x7fffedf998e0 <_nl_C_LC_CTYPE_class+256>, frame=0x7fffedf98fe0 <_nl_C_LC_CTYPE_toupper+512>, throwflag=0) at Python/generated_cases.c.h:1621
#15 0x00007ffff11fbd77 in _PyEval_EvalFrame (tstate=0x7ffff157a140 <_PyRuntime+315680>, frame=<optimized out>, throwflag=0) at ./Include/internal/pycore_ceval.h:120
#16 _PyEval_Vector (tstate=tstate@entry=0x7ffff157a140 <_PyRuntime+315680>, func=func@entry=0x7fffead81220, locals=locals@entry=0x7fffebaf4600, args=args@entry=0x0, argcount=argcount@entry=0, kwnames=kwnames@entry=0x0)
    at Python/ceval.c:2110
#17 0x00007ffff129eb84 in PyEval_EvalCode (co=0x7fff293e7b30, globals=<optimized out>, locals=0x7fffebaf4600) at Python/ceval.c:982
#18 0x00007ffff12c0d63 in run_mod (mod=mod@entry=0x55555a345b70, filename=filename@entry=0x7ffff1535978 <_PyRuntime+35160>, globals=globals@entry=0x7fffebaf4600, locals=locals@entry=0x7fffebaf4600, flags=flags@entry=0x0, 
    arena=arena@entry=0x7fff292f16d0, interactive_src=0x0, generate_new_source=0) at Python/pythonrun.c:1460
#19 0x00007ffff12c084a in _PyRun_StringFlagsWithName (str=<optimized out>, name=0x7ffff1535978 <_PyRuntime+35160>, start=<optimized out>, globals=0x7fffebaf4600, locals=0x7fffebaf4600, flags=0x0, generate_new_source=0)
    at Python/pythonrun.c:1260
#20 0x00007ffff2574e1a in Base::InterpreterSingleton::runString[abi:cxx11](char const*) (this=0x55555565f440, sCmd=0x555556202ed0 "FreeCAD.open(u\"/tmp/fc/crash-step-export.fcstd\")")
    at /media/source/FreeCAD/src/Base/Interpreter.cpp:291
#21 0x00007ffff5215f09 in Gui::Command::_runCommand (file=0x7ffff5e57358 "/media/source/FreeCAD/src/Gui/Application.cpp", line=855, eType=Gui::Command::App, 
    sCmd=0x555556202ed0 "FreeCAD.open(u\"/tmp/fc/crash-step-export.fcstd\")") at /media/source/FreeCAD/src/Gui/Command.cpp:827
#22 0x00007ffff5215aeb in Gui::Command::_doCommand (file=0x7ffff5e57358 "/media/source/FreeCAD/src/Gui/Application.cpp", line=855, eType=Gui::Command::App, sCmd=0x7ffff5e8ca5b "%s.open(u\"%s\")")
    at /media/source/FreeCAD/src/Gui/Command.cpp:772
#23 0x00007ffff4fda2b1 in Gui::Application::importFrom (this=0x7fffffffd120, FileName=0x55555a357150 "/tmp/fc/crash-step-export.fcstd", DocName=0x5555588a8f50 "Unnamed", Module=0x5555597980f0 "FreeCAD")
    at /media/source/FreeCAD/src/Gui/Application.cpp:855
[... cut ...]
#48 0x00007fffeaa94372 in ?? () from /usr/lib/qt6/plugins/platforms/../../../../lib/libQt6WaylandClient.so.6
#49 0x00007fffee5e88df in QObject::event (this=0x5555556dfb30, e=0x7fffdc001600) at ../src/corelib/kernel/qobject.cpp:1479
[... 42 lines cut ...]
<summary>BREP file</summary>

```
DBRep_DrawableShape

CASCADE Topology V1, (c) Matra-Datavision
Locations 0
Curve2ds 0
Curves 0
Polygon3D 0
PolygonOnTriangulations 0
Surfaces 0
Triangulations 0

TShapes 1
Co

1100000
*

+1 0
```

</details></li>
</ul>

flaviut, 2026-11-06:
Seems like all the files in github & the original freecad bug tracker are corrupted:

```
❯ unzip crash-step-export.fcstd.zip
Archive:  crash-step-export.fcstd.zip
FreeCAD Document
  inflating: Document.xml            
  inflating: PartShape.brp            bad CRC a13cfd03  (should be 61b11ba5)
  inflating: DiffuseColor            
  inflating: LineColorArray          
  inflating: PointColorArray
```

OCCT ought to not crash, just in general, but that would be a different bug.

The forum attachment is good though: https://forum.freecad.org/viewtopic.php?p=229740&sid=0f3bd96cdcbe6a23b49f005c7bb5b598#p229740

And STEP export succeeding just fine in 7.9.3, 8.0.1, and OCCT 3d097a0328

[image]

I've also confirmed that the file crashes on STEP export with OCCT 7.8.1:

```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff798b946 in TopoDSToStep_MakeStepFace::Init (this=this@entry=0x7ffffffc4970, aFace=..., aTool=..., FP=..., theLocalFactors=...) at ../source-7.8.1/src/TopoDSToStep/TopoDSToStep_MakeStepFace.cxx:374
374      C2d = Handle(Geom2d_Curve)::DownCast(C2d->Copy());
#0  0x00007ffff798b946 in TopoDSToStep_MakeStepFace::Init (this=this@entry=0x7ffffffc4970, aFace=..., aTool=..., FP=..., theLocalFactors=...) at ../source-7.8.1/src/TopoDSToStep/TopoDSToStep_MakeStepFace.cxx:374
#1  0x00007ffff79780fa in TopoDSToStep_Builder::Init (this=this@entry=0x7ffffffc4bb0, aShape=..., myTool=..., FP=..., theTessellatedGeomParam=theTessellatedGeomParam@entry=2, theLocalFactors=..., theProgress=...) at ../source-7.8.1/src/TopoDSToStep/TopoDSToStep_Builder.cxx:123
#2  0x00007ffff7979c46 in TopoDSToStep_Builder::TopoDSToStep_Builder (this=this@entry=0x7ffffffc4bb0, aShape=..., aTool=..., FP=..., theTessellatedGeomParam=2, theLocalFactors=..., theProgress=...) at ../source-7.8.1/src/TopoDSToStep/TopoDSToStep_Builder.cxx:61
#3  0x00007ffff7982499 in MakeManifoldSolidBrep (aShell=..., FP=..., theLocalFactors=..., theProgress=...) at ../source-7.8.1/src/TopoDSToStep/TopoDSToStep_MakeManifoldSolidBrep.cxx:54
```

Please let me know if you disagree about this being fixed, but I'm going to mark it as done for now.

closed 2026-11-06 by flaviut

luzpaz, 2026-11-11:
The OCCT LIason tag doesn't show what version this was supposedly fixed in, @pieterhijma

flaviut, 2026-11-11:
I've been doing a bit  of OCCT issue triage & testing here, sorry if I've failed to follow some part of the process. I added the related OCCT issue as the "blocked by" in the github sidebar, this was fixed with 7.9.