FreeCAD/FreeCAD #5785: a python method to convert  a Shape into a point cloud without extra dialog but parameter

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3446

* **Reporter:** microelly2
* **Date submitted:** 4/25/2018
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** Mesh

# Original report text

see 
https://forum.freecadweb.org/viewtopic.php?f=22&t=28366#p229963

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

Nicolaus93, 2022-04-04:
Any development on this?

luzpaz, 2022-04-07:
@Nicolaus93 we're currently in a feature freeze but you can bump the forum thread and see who weighs-in...?

Nicolaus93, 2022-04-11:
[This conversation](https://forum.freecadweb.org/viewtopic.php?p=229986&sid=9d09be58fdfbf90981b85440b3e0823e#p229986) is from almost 4 years ago. Should I post another reply on the same thread or do something else? 
[image]

luzpaz, 2022-04-11:
Yes, please  post to the thread and x-post back to this ticket (Since it's migrated to GH from MantisBugtracker)

PR #6745 (2022-06-21): Points: fixes #5785: a python method to convert a Shape into a point …
…cloud without extra dialog but parameter

Thank you for creating a pull request to contribute to FreeCAD! To ease integration, we ask you to conform to the following items. Pull requests which don't satisfy all the items below might be rejected. If you are in doubt with any of the items below, don't hesitate to ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10)!

- [ ]  Your pull request is confined strictly to a single module. That is, all the files changed by your pull request are either [... cut]
commit 8998c984 (2022-06-27): Points: fixes #5785: a python method to convert a Shape into a point cloud without extra dialog but parameter