FreeCAD/FreeCAD #5789: Control objects visibility via expressions and spreadsheets cells

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3466

* **Reporter:** silopolis
* **Date submitted:** 5/8/2018
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** expressions

# Original report text

Feature would allow to control object visibility via expressions and/or spreadsheets cells returning 0/1 or True/False.

# Additional information

This would be valuable in many cases as witnessed by users interest in the following threads:
* https://forum.freecadweb.org/viewtopic.php?f=3&t=18681
* https://forum.freecadweb.org/viewtopic.php?f=10&t=18065

AFAIC, in a woodworker/cabinetmaker use case, this feature would make my base parametric parts, cabinets/furnitures, assemblies much more powerful:
* cabinets with/without shelves/separations,...
* panels with/without edge banding, face veneering

For BoM/cutlist generation, an "Enabled" boolean property (even a "dumb" one) would be further empowering by allowing to manage visibility and "existence" independently.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS: Debian 9 "Stretch" Linux 4.9**
* **Platform:** Intel x86_64
* **Updated:** 2/6/2021

adrianinsaval, 2022-03-19:
this is actually possible AFAIK, via right click->show all and then right click->Expression in any case the request would be to have this normally available without the right click actions I guess.

adrianinsaval, 2022-03-19:
I think this issue was created before this was possible, so I think it can be closed now. This is pretty advanced usecase IMO so it's fine to have it behind show all and right click menu

closed 2022-03-19 by chennes

yomajo, 2023-10-20:
> this is actually possible AFAIK, via right click->show all and then right click->Expression in any case the request would be to have this normally available without the right click actions I guess.

Right click on what?

Then Right click on what?

How to make visibility parametric, control via configurations, variables or spreadsheet exactly?
[Screencast from 2023-10-20 15:29:02.webm](https://github.com/FreeCAD/FreeCAD/assets/45366313/75bbf414-1930-4123-9b4d-d53171f9bc74)

adrianinsaval, 2023-10-20:
https://github.com/FreeCAD/FreeCAD/assets/36372335/8c48225f-b14b-45ae-8d3f-adde6d0a9365

yomajo, 2023-10-21:
Thank you @adrianinsaval this was very useful! I'm looking for ways to control both: bodies and features within FreeCAD programatically.

Please see below:

Trying to hide fillet (show pad). Bodies work as expected. I can both hide and show them from python in resulting .stl, however features are acting up on me. I don't know if it's a bug, limitation of current architecture or expected behavior because of ...?

This is full demo:
[Screencast from 2023-10-21 13:26:09.webm](https://github.com/FreeCAD/FreeCAD/assets/45366313/44df5581-ad81-47b3-8c44-ab5030b6221a)

Final variable values of saved part:
[image]

Python script, stripped of rubbish:
```
import FreeCAD, Mesh
from my_utils import get_export_objs

MODEL_FPATH = 'cad_models/bodies_visibility.FCStd'
EXPORT_FPATH = 'exports/bodies_visibility_after.stl'

doc = FreeCAD.open(MODEL_FPATH)
vars = doc.getObject('Variables')

# hide fillet on b1, adjust l/w on b1
vars.b2vis = True
vars.b1filletvis = False
vars.b1padvis = True
vars.b1L1 = 100.0
vars.b1W1 = 50.0
vars.b1vis = True

TGT_TREE_BODIES = ['body_1']
if vars.b2vis == True:
    TGT_TREE_BODIES.append('body_2')
doc.recompute()
doc.save()

# export
export_bodies = get_export_objs(doc, TGT_TREE_BODIES, use_labels=True)
Mesh.export(export_bodies, EXPORT_FPATH)
FreeCAD.closeDocument(doc.Label)
```

adrianinsaval, 2023-10-21:
For detailed help requests please head to the forum