FreeCAD/FreeCAD #5792: Toggle visibility of all construction geometry

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3489

* **Reporter:** Rafa Mahiques
* **Date submitted:** 5/21/2018
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** 

# Original report text

First of all, congratulations for all the new features that have been added to the latest version. It has great capabilities now and makes the projects smoother.
I can think of many scenarios where this feature would be useful, but there is a specific one where I really miss it.
When 2 lines of different nature (normal, construction, external) overlap and you want to select the overlapped one to apply any constraint or function (fillet, scale, etc.) you have to find the instance in the "elements" list and select it there.
However, it would be useful that the overlapped geometry can be selected without having to identify it in that list. Let's say with a combination of a keystroke and the mouse wheel.

# Additional information

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6712 (Git)
Build type: Release
Branch: releases/FreeCAD-0-16
Hash: da2d364457257a7a8c6fb2137cea12c45becd71a
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2018-05-26 21:56
Rafa Mahiques please open a forum thread to discuss with community before opening tickets as per the enormous yellow banner at the top of this page. 

---

### Comment by Kunda1 2018-05-29 18:48
Rafa Mahiques, Please respond.

---

### Comment by normandc 2018-06-11 20:12
Moved to Sketcher as description seems to indicates this is specifically about it.

luzpaz, 2023-11-11:
Any relation to #5639 ?

maxwxyz, 2024-01-05:
> Any relation to #5639 ?

Guess so

maxwxyz, 2024-01-14:
@Roy-043  ^ also

closed 2024-01-14 by Roy-043