FreeCAD/FreeCAD #5799: fillet fails on cut torus

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3554

* **Reporter:** bzipitidoo
* **Date submitted:** 8/6/2018
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** feedback
* **Tags:** Part

# Original report text

Been using python scripting to make parts.  Fillet is finicky.  Sometimes it works.  Sometimes it gives an error message:

Part.OCCError: BRep_API: command not done

and the fillet fails.  It can fail on a simple piece, but work on a more complicated shape that is other shapes fused with the simple piece.

Investigating, the simplest fail I found so far is a cut down torus-- sliced in half like a bagel, then cut in half again.  See the included python script.

# Additional information

OS: "PCLinuxOS"
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13528 (Git) AppImage
Build type: None
Branch: releases/FreeCAD-0-17
Hash: 5c3f7bf8ec51e2c7187789f7edba71a7aa82a88b
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

# Steps to reproduce

```#makeFillet fails in this python script:

import math
import Part
from FreeCAD import Base

DEG2RAD = math.pi/180.0
rd = 72
origin = Base.Vector(0,0,0)
znorm = Base.Vector(0,0,1)
xnorm = Base.Vector(1,0,0)

boxcorner = Base.Vector(-rd*2,-rd*2,-rd*2)
cutbox = Part.makeBox(rd*4,rd*2,rd*4,boxcorner,znorm)

fulltorus = Part.makeTorus(rd-4,4,origin,xnorm,0,360)
halftorus = fulltorus.cut(cutbox)

basecyl = Part.makeCylinder(rd+2,9,origin,xnorm)
halftorus = halftorus.cut(basecyl)

#This fillet fails
filletht = halftorus.makeFillet(0.2, halftorus.Edges)```

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Linux kernel 4.11.8**
* **Platform:** PC
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2018-08-07 12:01
bzipitidoo please test this on 0.18dev as well (there is an [appimage](https://github.com/FreeCAD/FreeCAD/releases/tag/0.18_pre)). Also please, per our guidelines (yellow banner at the top) post to our forums before opening a ticket. 

---

### Comment by bzipitidoo 2018-08-07 13:29
I tested this script with makeFillet on version 0.18.14220 and 0.18_Conda Appimages.  The problem with makeFillet is present in both 0.18 versions.

---

### Comment by Kunda1 2018-08-07 15:36
please open a forum thread and post this ticket to it to start a discussion. thanks

---

### Comment by normandc 2018-09-02 21:47
For the Python challenged, an actual FreeCAD document would be nice. I'm an experienced CAD user, I can troubleshoot models, but I need a model to do it - a python script might as well be Mandarin to me. I don't speak Mandarin. And running the aforementioned script produces zero visible result in the 3D view.

---

### Comment by wmayer 2018-10-04 09:15
Attached is a project file with the half torus. Now go to Part, select the fillet tool and select all edges. Use a radius of 0.2.

---

### Comment by Kunda1 2018-11-07 00:51
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=32087

---

### Comment by normandc 2018-11-11 05:37
The problem seems to be with the inner circular edge of the half torus. Without it, all other edges fillet successfully.

A workaround is to cut the torus slightly bigger or smaller than half. If the minor radius of the torus is 4mm, cut 0.001 from the midplane, either side works - all fillets succeed.

I would surmise that this is an OCC issue...

OS: Ubuntu 18.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15169 (Git)
Build type: Release
Branch: master
Hash: 7d197ce829600e308dd0e85aeb1a4ac9376ff286
Python version: 2.7.15rc1
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/Canada (fr_CA)

luzpaz, 2022-05-28:
@wwmayer is this an OCC issue?

wwmayer, 2022-05-29:
Seems so.

luzpaz, 2022-05-29:
Added to [OCC bugs in the bugtracker](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264) thread

grubertechnic, 2023-07-29:
Stil an issue.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33658 (Git)
Build type: Release
Branch: master
Hash: ba5f7e6d75dee09ec7076ad24895b04e56f21b63
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

Note: for the torus the BOPcheck fails, even when made new fro scratch.

[image]

ElementW, 2026-08-24:
Still fails with `BRep_API: command not done` nowadays, on both `26.3.0dev.20260819` with OCC 7.8.1, and faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c with OCC 8.1.0.dev1-3d097a0328