FreeCAD/FreeCAD #5801: PATH: Facing Operations with large tools

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3570

* **Reporter:** dubstar_04
* **Date submitted:** 8/27/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** 

# Original report text

It would be awesome if facing operations would allow the use of tools larger than the stock and had the option to make a single pass facing op. 

This would allow the use fly cutter and facemills. 

# Steps to reproduce

none required

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2018-08-27 22:29
dubstar_04 please include the link to the forum post you mentioned this in before you opened a ticket. Thanks

---

### Comment by dubstar_04 2018-08-28 04:46
Forum post: https://forum.freecadweb.org/viewtopic.php?f=15&t=30525

sliptonic, 2022-03-21:
Bumping this to 0.21.  Let's get serious about it and discuss.
Technically, it's possible to achieve this now with either the pocket operation + extensions or the slot operation.  Neither of those is a very good fit.
In my mind, a facing operation is done on the top of the stock to achieve a uniform starting condition.  Should mill face be adjusted with that assumption? 
For example,  I could see it having now boundary setting at all.  Instead, the boundary would be determined from the stock perimeter + the tool diameter. The operation would never go deeper than the top of the model.

sliptonic, 2024-05-07:
see also 
#10585
#8272
#6849
#11410

AdrianCM95, 2024-11-06:
I have a feeling this operation is not thought to use a flycutter or a facemill (like the one in the picture). However, i think this should be the primary purpose of this operation.

To my knowledge, during a facing operation, the tool should cut only when doing XY movements.
[image]

In order to have a similar behavior as one of this facemills is used, I need to set the "material allowance" to at least the tool diameter. In my case, 60mm. But then you start getting some weird paths.

- There is additional movement which is not needed.
- Circular turns would be also nice to have.
- If allowance is not high enough, the tool would plunge from the top as if it could cut in the middle.

**Zigzag 90º angle, 60mm allowance:** 
[image]
desired behavior:
[image]

**Zigzag 0º angle, 60mm allowance:** Same as above.
[image]
desired behavior:
[image]

Just for reference i have listed this examples using line

**Line 90º angle, 60mm allowance:** 
[image]

**Line 90º angle, 30mm allowance:** 
[image]

Finally, in regards to the allowance parameter, having X mm of allowance doesn't mean you need to use it on every direction.
Maybe a "directional additional travel" would be more useful.
[image]

I hope this comment is useful and helps to clarify the desired behavior some of us expect from this tool.
For further clarification, just contact me, i am willing to support.