FreeCAD/FreeCAD #5802: TechDraw: Easier Color Drawings

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3572

* **Reporter:** Kunda1
* **Date submitted:** 8/31/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** #pending-forum

# Original report text

Pretty self-explanatory
Reference: https://forum.freecadweb.org/viewtopic.php?f=3&t=30524&p=253695#p253670

# Other bug information

* **Priority:** none
* **Severity:** feature
* **Category:** Feature
* **Updated:** 1/19/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2022-01-19 17:53
forum thread bumped https://forum.freecadweb.org/viewtopic.php?p=563252#p563252

luzpaz, 2023-09-25:
@WandererFan what does this ticket need, if anything?

WandererFan, 2023-09-25:
You can change the colour of any line in a TechDraw View and I don't think that was possible when the issue was created.  

I'm going to close this and wait for somebody to file one that asks for the colours from the 3d window to be automagically transfered to the TD page.

closed 2023-09-25 by WandererFan