FreeCAD/FreeCAD #5803: TechDraw: Don't recompute on text changes

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3575

* **Reporter:** Kunda1
* **Date submitted:** 8/31/2018
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** #pending

# Original report text

Reference: https://forum.freecadweb.org/viewtopic.php?f=3&t=30524&p=253695#p253670

# Other bug information

* **Priority:** none
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/2/2022

# Discussion from Mantis ticket

### Comment by wandererfan 2018-10-05 12:46
I don't understand this one.  What text is being changed?   Is Page KeepUpToDate property set to false to avoid recompute/redraw?

---

### Comment by jango 2019-02-21 04:39
The problem is the following: when we're making a technical draw of some machine, we need to specify a lot of components characteristics and names in order to make it easily understandable; in example the table of components, welding type, surface finishing and so forth; in the designing process we change continuously those characteristics and text, in this sense is annoying have to recompute a whole text if you only need to change a section of it: an incorrect letter or number, recompute a whole 50 components table for only one incorrect part. Try to make those texts individual boxes, like Inventor.
!(https://proxy.duckduckgo.com/iu/?u=https%3A%2F%2Ftse3.mm.bing.net%2Fth%3Fid%3DOIP.M-30Hqj1UuH3McNBCqQkJwHaFP%26pid%3D15.1&f=1)

---

### Comment by wandererfan 2019-02-23 13:36
This may be a SpreadsheetWB issue.  See [forum post](https://www.forum.freecadweb.org/viewtopic.php?f=35&t=30524&start=10#p288350)

---

### Comment by Kunda1 2019-03-03 14:09
wmayer can you weigh in ?

---

### Comment by wmayer 2019-03-04 09:24
Is there an example project available to test the behaviour?

---

### Comment by wandererfan 2019-03-04 13:55
Here is the one I've been using. Still don't know what text the OP is changing.

---

### Comment by wandererfan 2019-07-19 15:26
How can this be "confirmed" if we don't know what the OP is talking about?

---

### Comment by Kunda1 2019-07-19 19:56
wandererfan hmm, looks like I did that. Not sure what my rationale was at the time. Apologies.

---

### Comment by Kunda1 2019-07-19 19:57
wandererfan though, instead as marking it as 'new' better to mark as 'feedback' because we are awaiting feedback from OP in that case

---

### Comment by wandererfan 2019-07-20 00:16
kunda1 - thought I did make it "feedback".  getting old.

---

### Comment by Kunda1 2022-01-19 17:46
jango ping

---

### Comment by wandererfan 2022-02-02 17:07
The forum discussion is marked as "Solved".  Can we close this ticket?

Roy-043, 2023-07-16:
Closing as stale.

luzpaz, 2023-09-25:
Tragically the original OP never responded and the uploaded example file didn't 'survive' in the old mantisBT database. I've edited the bug to show the original images and content a little better. But since @WandererFan couldn't reproduce without OPs input + deduced this is a Spreadsheet issue...yes. it's best to close this ticket.

closed 2023-09-25 by luzpaz