FreeCAD/FreeCAD #5804: Orthogonally restrict dragging of geometry in Sketcher (invoked by pressing ctrl or shift or ctrl+shift)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3576

* **Reporter:** ceremcem
* **Date submitted:** 8/31/2018
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** shortcuts

# Original report text

It would be very helpful to have a mode to restrict dragging of lines, vertices or possibly other geometry to only one axis (vertical or horizontal) while creating a sketch. 

My first reflex was pressing Ctrl or Shift or Ctrl+Shift keys while dragging a rectangle edge, just like is done in Gimp.

Here is the forum link we discussed the issue: https://forum.freecadweb.org/viewtopic.php?f=3&t=30501

# Steps to reproduce

(Can be seen in the forum thread)

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by normandc 2018-09-01 15:36
Reworded summary description to better convey meaning.

maxwxyz, 2024-03-30:
Not sure if this is still relevant as the sketch should be constrained anyways. There is a vertical or horizontal constraint and the move tool also lets you transform it precisely.
If there is no other feedback I will close this soon.

ceremcem, 2024-04-01:
I'm not using FreeCAD for a long time now, so we may better close it since it doesn't make sense to you after that time.

closed 2024-04-01 by maxwxyz