FreeCAD/FreeCAD #5809: Fillets with variable radius

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3608

* **Reporter:** chrisb
* **Date submitted:** 9/23/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

It would be great to have fillets with variable radius in Part and PartDesign workbenches. See [german forum entries](https://forum.freecadweb.org/viewtopic.php?f=13&t=26646&p=258035#p257860), Werner is already involved.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by wmayer 2020-09-13 09:45
https://forum.freecadweb.org/viewtopic.php?f=13&t=50170

app4soft, 2023-07-05:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/5723

luzpaz, 2023-07-09:
Closing in favor of #5723  
(thanks)

closed 2023-07-09 by luzpaz