FreeCAD/FreeCAD #5810: TechDraw - Auxiliary View

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3609

* **Reporter:** normandc
* **Date submitted:** 9/24/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Per Wikipedia:
[quote=Wikipedia]An auxiliary view is an orthographic view that is projected into any plane other than one of the six primary views. These views are typically used when an object contains some sort of inclined plane. Using the auxiliary view allows for that inclined plane (and any other significant features) to be projected in their true size and shape. The true size and shape of any feature in an engineering drawing can only be known when the Line of Sight (LOS) is perpendicular to the plane being referenced.[/quote]
https://en.wikipedia.org/wiki/Engineering_drawing#Auxiliary_views

The auxiliary view has a "viewing plane indicator" (really no clue how it's called) which usually is a dotted line with arrows, similar to a section view. It may depend on drawing standards. The auxiliary view is usally linked and aligned to its parent view, although it may be possible to break the alignment if there is not enough space on the page besides the parent view.

A possible work flow would be to select a linear edge on a view then launch the auxiliary view tool. A FreeCAD file with TechDraw page is attached, with a pseudo auxiliary view made with the standard NewView tool. Attached svg was edited to manually add the plane indicator with view label.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

WandererFan, 2022-09-21:
[image]
[TechDraw_Feature_Request_Auxiliary_View_Example_nc1.fcstd.zip](https://github.com/FreeCAD/FreeCAD/files/9611927/TechDraw_Feature_Request_Auxiliary_View_Example_nc1.fcstd.zip)

WandererFan, 2022-12-09:
Additional information: 
https://wiki.freecadweb.org/TechDraw_SectionExamples#Auxiliary_view
https://www.forum.freecadweb.org/viewtopic.php?p=646546#p646546

WandererFan, 2022-12-09:
Description of the Work

The DocumentObject DrawAuxiliaryView inherits DrawViewPart, with some added properties:

- BaseView - the DrawViewPart that this DAV relates to
- AuxiliaryDirection  - projection direction relative to BaseView. Used to calculate the absolute projection direction and rotation.
- ReferenceLabel - the text that appears in both the BaseView and the DAV.

A new graphics object (ex. QGIViewAuxiliary) is not required as the painting of the view is the same as a regular view.

QGIViewPart will need to be modified to draw the reference Text and arrow on top of the BaseView.

A graphics object for the Reference and Arrow will probably be needed.

The existing ViewProvider for DVP can likely be used, but need to confirm that changes to AuxiliaryDirection and ReferenceLabel can be detected in ViewProviderViewPart. If necessary, a light weight view provider can be derived as is done for sections.

A new command will need to be added to TechDraw/Gui/Command.cpp and TechDraw/Gui/Workbench.cpp.

A new task dialog will be needed to set direction. 

A new wiki page will be needed. 

Refer to the checklist for incidental tasks.
[newFeatureChecklist.txt](https://github.com/FreeCAD/FreeCAD/files/10198691/newFeatureChecklist.txt)

pierreporte, 2025-04-29:
A **150 € reward** will be given to the contributor that writes a pull request, solving this issue, that gets merged. It will be paid by the FreeCAD Project Association. It is to be considered an incentive. Grant applications are welcome, but the total amount will exclude the reward.

FC-FBXL5, 2025-05-05:
> A new task dialog will be needed to set direction (see TechDraw/Gui/TaskSectionView.cpp).

Please reconsider. The direction should be set graphically by selecting vertices or straight edges in the base view. This resembles the workflow in descriptive geometry done manually that has been used for ages. Other CAD apps don't do it this way without reason. (this applies to section views, too)

A task dialog should only have a checkbox to reverse the direction.

FC-FBXL5, 2025-05-08:
I'm looking forward to an auxiliary view feature integrated in TD and prepared a preliminary wiki page containing a proof of concept macro in Python.
I chose this way as I'm still unable to compile C++ code. Maybe this helps someone in the meantime while waiting for the one to pick the bounty. See https://wiki.freecad.org/TechDraw_AuxiliaryView

franklincg, 2026-08-17:
Hi, I'm interested in working on this bounty. Could you confirm the current reward amount and whether the issue is still available for a new contributor to work on?