FreeCAD/FreeCAD #5814: FEM, calculix, support analysis type 'no analysis' for model checking

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3622

* **Reporter:** berndhahnebach
* **Date submitted:** 10/3/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

https://forum.freecadweb.org/viewtopic.php?f=18&t=31303#p260032

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by berndhahnebach 2019-03-01 15:23
https://github.com/FreeCAD/FreeCAD/commit/40d0db3

but it needs to be added to the task panel too.

FEA-eng, 2023-12-04:
This issue can be closed as completed:

[image]

FEA-eng, 2024-02-19:
@maxwxyz Can you close this as completed (screenshot above shows it in the task panel)?

maxwxyz, 2024-02-19:
Closed as implemented.

closed 2024-02-19 by maxwxyz