FreeCAD/FreeCAD #5819: At FreeCAD startup; Add options to change the initial camera view and window size in Sketcher.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3637

* **Reporter:** freedman
* **Date submitted:** 10/12/2018
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Add options for Sketcher to use when FreeCAD initially starts and no file has been opened.

https://forum.freecadweb.org/viewtopic.php?f=8&t=31000

***To summerize***, there are two issues, both with starting a sketch for the first time.
1) This text from a post stated well: 
"offer a setting in Preferences --> Display to set the initial view to one of the standard views (top, right, front, left, axonometric...), but change the default from top to axonometric."

2) This text from a post stated well:
" the zoom is waaay to close. The whole width of the screen is about 2 mm. Suitable for designing the tiniest of parts". My screen measures 2.5 x 1.5, very small also.

#1 above could be very helpful to the new user, they would get a much better idea of the 3D workspace.
Thanks

# Additional information

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14786 (Git)
Build type: Release
Branch: master
Hash: 75acfe898f97d213647b0b93032287e48a31abfe
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by freedman 2018-10-12 17:26
The first line of the description should say "Add options for Sketcher to use when FreeCAD initially starts and no file has been opened.

---

### Comment by Kunda1 2018-10-13 12:27
freedman added

maxwxyz, 2024-03-24:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12895

maxwxyz, 2024-03-24:
I'm closing this in favor of #12895 as there is a better demo

closed 2024-03-24 by maxwxyz