FreeCAD/FreeCAD #5824: GeomAPI_ExtremaCurveCurve fails to detect intersection (upstream OCCT bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3655

* **Reporter:** Kunda1
* **Date submitted:** 10/28/2018
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** #ready-for-migration, OCC Bug, OCC NotOnRoadmap, upstream

# Original report text

This ticket is for tracking https://tracker.dev.opencascade.org/view.php?id=30316
Original forum thread: https://forum.freecadweb.org/viewtopic.php?p=265166#p265166

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Bug
* **Updated:** 1/24/2022

luzpaz, 2022-05-27:
Looks like upstream merged the fix. It's should be in OCC 7.6.2   
Testers needed

maxwxyz, 2024-04-06:
Is this still relevant or already fixed?

maxwxyz, 2024-11-09:
Closing this for no feedback

luzpaz, 2024-11-09:
This is an OCC bug. I can't test ATM. @AIRCAP do you mind quickly checking to see if this is resolved?

luzpaz, 2026-09-20:
Needs to be tested against the attachments in the forum thread: https://forum.freecad.org/viewtopic.php?p=263524

flaviut, 2026-11-06:
Fixed in OCCT 8.1.0.dev1 https://github.com/Open-Cascade-SAS/OCCT/commit/53706f78a8640c4407bdcc6752df13a270b159f2

[issue-5824-macro.zip](https://github.com/user-attachments/files/34488263/issue-5824-macro.zip)
[image]

Fix: https://github.com/Open-Cascade-SAS/OCCT/pull/1634

flaviut, 2026-11-06:
Fixed in OCCT 8.1.0