FreeCAD/FreeCAD #5826: Visibility not preserved after making edits in building tree

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3664

* **Reporter:** djmccorrie
* **Date submitted:** 11/1/2018
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** assigned
* **Tags:** BIM, Visibility

# Original report text

Using the BIM workbench, if you have several floors as building parts within a building. Hide the floors your not working on. Make an edit, such as moving an object, changing the width of a wall. The whole building becomes visible and have to go through hiding the levels not working on again.

This seems to be a general thing, after any edit the visibilities are lost... making it quite frustrating if you need to make a few edits to a layer obscured by others. Another scenario where the same issue exists is where you need to hide a wall to get a good side profile of an internal wall. Same happens again... keep having to hide the outer wall on every edit.

# Additional information

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15113 (Git)
Build type: Release
Branch: master
Hash: e86f7caea866db5b78cfa78e5d3a10b735b205f9
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedKingdom (en_GB)

# Steps to reproduce

1.  open FreeCAD
2.  open BIM workbench
3.  Create Building
4.  Create 2x Building Parts
5.  Create 2x Wall
6.  Put one wall in each building part and the building parts in the building
7.  Hide a building part
8.  Edit the width of the wall in the other building part
9.  Both Walls visible

# Other bug information

* **Priority:** high
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by djmccorrie 2018-11-01 07:36
Apologies If I've categorised this wrong, I'm new to this. Fairly sure this is a bug, rather than a feature...

---

### Comment by djmccorrie 2018-11-01 07:51
Found a workaround... it seems this only affects the visabilities with the same parent. So if I move the Ifc floor (building part) out from under the Ifc building. Hide the building and make edits to the floor in the root of the tree, the building object stays hidden. When the series of edits are finished I can move the building part back under the building.

This will remove some of the frustration, and allow faster work flow.

btw, thank you for your effort in putting this excellent program together. I am a developer, so when I've got some free time I might have a look over the source code and see if I can help.

---

### Comment by yorik 2018-11-01 20:48
Doing some tests here, the behaviour still happens if you use 2 groups instead of the 2 buildingParts, but not if the host is a group too instead of a Building.

If you place a third object directly inside the building, the bug happens with it too. 

So the problem seems to be that the host building (which is actually also a building part) forces the visibility of its children on/off when you change a property of one of its children... This is weird. There is nothing related to visibility inside the BuildingPart code, but there might be a problem with the Gui::ViewProviderGroupExtensionPython...

Investigating further.

---

### Comment by Roy_043 2019-09-13 09:31
This V0.19 issue seems related:
https://forum.freecadweb.org/viewtopic.php?f=3&t=39255

luzpaz, 2022-03-15:
@yorikvanhavre should this be moved to BIM WB issue queue?

luzpaz, 2022-09-13:
Bumped the thread suggested by Roy at https://forum.freecadweb.org/viewtopic.php?p=625971#p625971  
CC @Roy-043 (just linking you to the issue (since you've commented on it in the past)

Roy-043, 2023-01-01:
I cannot reproduce the issue in v0.20.2 and v0.21.0. Therefore closing this.
```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31432 (Git)
Build type: Release
Branch: master
Hash: 63a2855751d638bf27ed969dc60654bda2b54baf
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

closed 2023-01-01 by luzpaz