FreeCAD/FreeCAD #5829: Thickness operation for part fails (upstream OCC bug)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3670

* **Reporter:** 0kko
* **Date submitted:** 11/5/2018
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #ready-for-migration, OCC Bug, OCC NotOnRoadmap, upstream

# Original report text

Thickness operation in Part and Part Design -workbences fails with: "BRep_API: command not done."

# Additional information

Binaries tested with "case.fcstd"

FreeCAD-0.17.13541.9948ee4.glibc2.17-x86_64.AppImage
FreeCAD-0.17.13541.9948ee4.glibc2.17-x86_64.AppImage

Note: Issue is reproducible on latest v0.19 since the bug is with upstream OCCT
This ticket is being tracked via [OCC Bugs in the Bugtracker](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264) forum thread

# Steps to reproduce

Select "face 8" in "case.fcstd".
Try to do "Thickness"

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2018-11-05 18:37
0kko Please check with 0.18_pre appImage to see if you can recreate. https://github.com/FreeCAD/FreeCAD/releases/tag/0.18_pre
(make sure you backup you file before doing so as 0.18 is not backward compatible if you open said file in 0.17 again)

BTW, your ticket does not follow our guidelines stated in the enormous banner at the top of this page. We ask for people to report their issues on the forum first, many times they don't even make it to the tracker because they either get solved, or there is an open ticket already (there are exceptions). You also tested this on 0.17 yet you labeled this 0.18. + we have no idea what OS or OS version you are running. Guidelines are there for a reason, thank you.

---

### Comment by normandc 2018-11-11 05:58
It is no surprise that the Thickness operation fails - look at the corners at the joint between the wedge and the box. The fillet created a deep cavity at the four corners.

While this may look like a separate and legitimate bug (a fillet causing a crease between slanted corners), fixing it does not solve the thickness issue. The thickness command is notoriously weak. It is most probably an upstream (OCC) issue over which the FreeCAD team has no control.

It is indeed exasperating that some people are completely blind to the f***ing huge yellow banner at the top. IMO we should start ruthlessly closing reports that do not comply with rules. Enough with this.

---

### Comment by chrisb 2020-12-22 00:33
File is lost, and it was as Normand guessed most probably an OCC issue anyway.

---

### Comment by wmayer 2020-12-22 11:53
https://forum.freecadweb.org/viewtopic.php?f=8&t=53514&p=460091#p460091

---

### Comment by Kunda1 2020-12-22 12:13
Reopening and tracking via [OCC Bugs in the Bugtracker](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264) forum thread

---

### Comment by Kunda1 2020-12-22 12:15
Re-uploaded attachment since previous one was broken. Thanks wmayer ;)

---

### Comment by Kunda1 2022-01-23 00:34
I can't find face8 in the attached file ???
chrisb care to check?

pieterhijma, 2025-03-13:
I can (still) reproduce this:

```
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.40717 (Git)
Build date: 2025/03/13 13:32:18
Build type: Release
Branch: main
Hash: a375752aa4ea4d416f757299c716820546fd41cc
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
```

Although it is clearly something that does not work as you would expect, in my humble opinion this particular example goes against the common practice that fillets and chamfers are applied as a final step in the design.

Moreover, if you do the Thickness operation first and then the fillets, then it works as expected:

[image]

At the least I think this issue should not have a high priority.

codeotaku, 2026-08-31:
Looking into this.  At first I thought this might be a fillet issue that later prevented the thickness operation, which I encountered recently.  But, this might actually be a case where FreeCAD is passing global (rather than local) coords to OCC... passing in local coords results in correct geometry. It's effectively a one line change. 

[image]

Will submit a PR shortly.

@pieterhijma perhaps not an OCC issue on this one?

closed 2026-09-01 by maxwxyz