FreeCAD/FreeCAD #5830: Sketcher Analysis interface improvements

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3671

* **Reporter:** abdullah
* **Date submitted:** 11/5/2018
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Werner's comments on PR #1554:

It's indeed a bit odd to have many new methods added to SketchObject just to have a convenient Python interface. Since the "analyser" member doesn't do anything else and to keep things a bit cleaner you can:

    keep the functions of SketchObjectPy but create each time a new instance of SketchAnalysis when one of these Python functions is invoked. This way you can remove the analyser member from SketchObject and all the methods. Also, this reduces possible side-effects in case of consecutive calls of validation functions unless you explicitly need this. But then it suffices to add a single method to SketchObject to return the SketchAnalysis instance.

    instead of putting the validation functions to SketchObjectPy you can create the class SketchAnalysisPy. In the long term this might be the better option since many new methods could be added. In order to invoke the validation functions from Python we have to add a single method to SketchObject(Py) to create and return a SketchAnalysis instance.

Nevertheless, I am going to merge the PR now and things can still be changed at a later time.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

maxwxyz, 2024-03-30:
@wwmayer is this still relevant?

maxwxyz, 2024-07-14:
@PaddleStroke do you think this is still relevant? Otherwise I'll close this for no feedback since 2022.

PaddleStroke, 2024-07-15:
I cannot understand the context/issue clearly here so I cannot say.

luzpaz, 2025-05-10:
Maybe @wwmayer can weigh-in for clarification ?