FreeCAD/FreeCAD #5831: HIDPI related bug where text doesn't fit grey box in "Documents" when the software starts.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3673

* **Reporter:** deejay2
* **Date submitted:** 11/7/2018
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** confirmed
* **Tags:** HiDPI

# Original report text

HIDPI is more and more used with > 1080p monitors like 4K. It basically downscales the resolution without actually lowering the monitor resolution. It results in crisper image without the tinyzation of everything on a small but high resolution monitor. 

Best explained here:
https://support.system76.com/articles/hidpi-multi-monitor/

QT5 actually supports HIDPI quite well and it will make FreeCAD usable on workstations using HIDPI (at last!) :) The problem(s) that remain(s) (Maybe I'll find more later) is in the Start page "Document" tab. Each thumbnail is in a gray box with text underneath. The text doesn't fit in the grey box so it makes the welcome to FreeCAD exeperience a bit frightening.

# Steps to reproduce

Use hidpi with a 4K monitor. For me it's easy to do with POP!_OS which derivates from Ubuntu. You just go in gnome settings->Peripherals->Monitors, Enable HIDPI and scale to 200%.

I used the FreeCAD_pre AppImage.

# Other bug information

* **Priority:** normal
* **Severity:** tweak
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by deejay2 2018-11-07 20:03
OS: Pop!_OS 18.04 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15150 (Git) AppImage
Build type: Release
Branch: master
Hash: fe8f172ce1f725cbc5e7a1538beaaac6a6529ec2
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/Canada (fr_CA)

---

### Comment by Kunda1 2018-11-09 00:17
deejay please open a thread on the forums to discuss this with the community 

---

### Comment by Kunda1 2018-11-10 16:07
forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=32159

---

### Comment by deejay2 2019-03-03 03:12
Here's another screenshot showing effect of hidpi on preferences.

---

### Comment by gweb 2019-04-06 19:13
This isn't only restricted to "documents". Here is a screenshot of my start page with a HiDPI scaling of 250% on V0.17 on Windows. Notice that not only the text doesn't fit in the boxes, but also the background is not stretched to fit the window

---

### Comment by gweb 2019-04-06 19:46
A few more instances of places where text or images don't fit correctly with HiDPI scaling

---

### Comment by Kunda1 2019-04-06 19:51
gweb we aren't addressing 0.17 anymore. Please test on 0.19dev 
https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre

---

### Comment by Kunda1 2019-06-17 07:32
Still lots of HiDPI bugs: https://www.freecadweb.org/tracker/search.php?tag_string=HiDPI

luzpaz, 2022-05-09:
Old Related: #5850 #6066

adrianinsaval, 2023-05-04:
is this still a problem? the report is pretty old

closed 2024-12-28 by github-actions[bot]