FreeCAD/FreeCAD #5832: Pocket operations in Spiral pattern sometimes leave island on face

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3676

* **Reporter:** gpothier
* **Date submitted:** 11/9/2018
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** new
* **Tags:** Path

# Original report text

Under some conditions, pocket operations leave islands in the middle of the face (see attached screenshots). Using the zig-zag pattern instead of spiral works around the problem.
This has been reported on the forums here: https://forum.freecadweb.org/viewtopic.php?f=15&t=23763
This is 100% reproducible on a freshly installed and updated Ubuntu 18.04.1 and FreeCad daily from the PPA repository.

# Additional information

OS: Ubuntu 18.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15162 (Git)
Build type: Release
Branch: master
Hash: 1d72ee43435fa099ae97a2f6e3d15a225d4f32d3
Python version: 2.7.15rc1
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

# Steps to reproduce

- Open the attached Latch.fcstd file
- In Fab / UpperBodyJob / Operations001, switch between Spiral and Zig-zag patterns.
- Observe that with Spiral pattern, there is a rather large island in the middle of the face, while with Zig-zag pattern the face is milled completely.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by normandc 2018-11-11 05:22
Fixed typo in summary

sliptonic, 2022-03-21:
Spiral pattern is buggy.  It's also largely unneeded since adaptive works the same way and much better.  
I'm proposing to disable spiral and close this bug.

luzpaz, 2022-03-21:
CC @mlampert

PR #6649 (2022-03-23): [Path] Fixes #5832
Remove unused and confusing spiral and triangle patterns

Thank you for creating a pull request to contribute to FreeCAD! To ease integration, we ask you to conform to the following items. Pull requests which don't satisfy all the items below might be rejected. If you are in doubt with any of the items below, don't hesitate to ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10)!

- [ ]  Your pull request is confined strictly to a single module. That is, all the files changed by your pull request are either in `App`, `Base`, `Gui` or one of th [... cut]
commit 05a8b67d (2022-03-23): Merge pull request #6649 from sliptonic/bug/5832

[Path] Fixes #5832
commit 5496accd (2022-03-23): [Path] Fixes #5832
Remove unused and confusing spiral and triangle patterns
closed 2022-03-23 by sliptonic