FreeCAD/FreeCAD #5835: Support for trimming arcs of hyperbola and arcs of parabola

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3684

* **Reporter:** abdullah
* **Date submitted:** 11/12/2018
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

It is not supported yet, so support it.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

Roy-043, 2023-07-16:
This has been implemented in V0.21 AFAICT. OK to close this one?

luzpaz, 2023-07-19:
@AjinkyaDahale ping (since abdullah is AFK)

AjinkyaDahale, 2023-07-19:
@luzpaz I had mentioned in #6971 that this is fixed. There is a difference between the splitting in that PR and the "trimming" that this issue requests, but as far as I know trimming was already done in at least by the time 0.20 was released.

closed 2023-07-19 by luzpaz