FreeCAD/FreeCAD #5836: Add Centermark for Circular BSplines

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3692

* **Reporter:** wandererfan
* **Date submitted:** 11/14/2018
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

TechDraw contains code to allow diameter and radius Dimensions for circular BSplines. 

Circular BSplines do not get a centermark like true circles do.   If the BSpline can be dimensioned, a centermark should be shown too. 

# Additional information

[Forum discusion.](https://forum.freecadweb.org/viewtopic.php?f=35&t=32235&p=268866#p268847)

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

grubertechnic, 2023-05-15:
@WandererFan : works on master

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33184 (Git)
Build type: Release
Branch: master
Hash: c33588be3e9e2bf555b133186a974d7dbd0b48f6
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

[image]

added the file here: [test.zip](https://github.com/FreeCAD/FreeCAD/files/11476745/test.zip)

Closing?

WandererFan, 2023-05-15:
closed as completed.

closed 2023-05-15 by WandererFan