FreeCAD/FreeCAD #5837: Constraints lose values upon reloading file.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3696

* **Reporter:** TheRainHarvester
* **Date submitted:** 11/16/2018
* **FreeCAD version:** 0.18
* **Category:** General
* **Status:** new
* **Tags:** 

# Original report text

When you use the "formula button" when typing in a constraint, it works only if you don't reload the file!  When reloading, the constraint reverts to the value that was saved.  It doesn't follow the formula anymore, so if you change a parameter, the linked constraint does not get updated.

# Additional information

OS: Windows 8.1
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15214 (Git)
Build type: Release
Branch: master
Hash: b3721f8613b975be13e1801f499fa06f6f0ad836
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

# Steps to reproduce

Open the attached file. Open the Sketcher window.
Notice the 44mm dimention to the VERY right.  it has the label "Global".
Notice the 44 mm dimension next to it (just left). We'll call this "LinkedVersion".  I saved it's formula's constraint as "Constraints.Global" but when you load the program, it gets set to the 44mm value.

You can repeat the bug by updating the "LinkedVersion" constraint by clicking the blue formula icon and typing "Constraints.Global".  

Now change the "Global" value (the length constraint to the VERY right).
The "LinkedVersion" will update as it should.
Now save the file. Close. And reload the file.

Now change the "Global" value (the length constraint to the Very right).
Notice that the "LinkedVersion" will not get updated!!

This bug can hide itself and ruin CNC paths if one is not aware of the problem.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** General
* **OS: 8.1 pro**
* **Platform:** win
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by abdullah 2018-11-16 19:09
Normandc
Kunda1

Can somebody from the CAD experts check this case to confirm?

---

### Comment by normandc 2018-11-17 16:40
IMO this should have been discussed in the forum first before a report was created; just as is **already** explained in the GIGANTIC yellow banner at the top.

This file document has multiple user errors. One not pertaining to this issue is starting a sketch in a Body then going to Part to create an Extrude. This produces the following warning in the Report view after a recompute:

```Part::Extrusion: Links go out of the allowed scope
```

One other thing hinting at an error is that at load time, the Report view prints the error "Failed to parse expression".

[quote=TheRainHarvester]Notice the 44 mm dimension next to it (just left). We'll call this "LinkedVersion".[/quote]
It is NOT called "LinkedVersion" in your file, its label is "Edge Vert 1 of 2". I highly doubt that spaces are allowed in contraint names. When I try to refer through it in an expression, it fails with "Failed to parse expression".

This file is a very poor case for a bug, and should have been simplified to only contain the elements to reproduce the issue. The whole Path job is irrelevant here. BTW it is completely inefficient to create such a sketch with >225 constraints and repeated circles. Posting on the forum before creating this bug, you would have received this advice.

I recommend that the next steps should be for the reporter to:
1. Register to the forum and create a topic to discuss the issue.
2. Following discussion and insight from forum regulars, provide a streamlined FreeCAD document that follows recommended practice, uses proper labels and still reproduces the issue.

---

### Comment by normandc 2018-11-17 16:41
As per comment above, please post on the forum for advice. There may be usage errors that are actually resulting in the issue.

---

### Comment by abdullah 2018-11-17 19:46
Thank you very much Normand. It saves me time when things are streamlined :)

---

### Comment by TheRainHarvester 2018-11-26 15:20
I added a simple file and started a discussion at:
https://forum.freecadweb.org/viewtopic.php?f=8&t=32437

---

### Comment by abdullah 2018-11-27 14:28
Thank you everybody for taking the time to streamline this.

I have moved this ticket to "Expressions", as the problem lies in the Expression Engine. It is not specific to the Sketcher.

Almost 40 years after the boom of the personal computer, spaces keep being a problem...

---

### Comment by Kunda1 2019-02-25 13:35
eivindkvedalen care to take a look at this?

adrianinsaval, 2022-03-16:
conflicting report, the created forum tracker talks about a constrain name with a space, that would be a duplicate of #5983 and #6593 the original report did not mention spaces but I can't access the original file in the bugtracker (I get a network error every time) to confirm but spaces in the name were mentioned by normandc so it was probably actually that. I think this can be closed.

adrianinsaval, 2022-03-16:
ok I get now what the original problem was, if you set a expression on a constrain and then give that constrain a name with a space it fails to parse the expression, even if the expression is entirely valid.
[test.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/8274703/test.FCStd.zip)

adrianinsaval, 2022-03-16:
Turns out there was already an issue that turned out to be this #5750 I guess this one can be closed

Roy-043, 2023-07-16:
Closing as duplicate.

closed 2023-07-16 by adrianinsaval