FreeCAD/FreeCAD #5839: Add an Offset tool to the Sketcher

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3708

* **Reporter:** normandc
* **Date submitted:** 11/24/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Offset is an essential tool of sketching. The offset tool should minimally work on lines, arcs, circles, and polylines made of lines and arcs. Making it work on complex curves (conics, B-splines) may prove difficult and be put aside for the initial implementation. Geometric constraints should be automatically added (coincident constraints for arcs, parallel constraints for lines, add appropriate constraints for polylines), as well as a distance constraint defining the offset value.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by abdullah 2018-11-27 14:33
Offset curves will be implemented for all existing geometry in v0.19. I am uncertain ATM how to treat constraints though. I will definitely start a thread on this at the beginning of the v0.19 development cycle.

---

### Comment by abdullah 2018-11-27 14:38
I realise now that you refer to the constraints making the offset. We do not need those constraints. An offset curve, which exist in OCC, is defined as a point on curve A plus a vector perpendicular to the curve on the plane (there is a vector product involved with the normal of the plane). Something like: Po(u)=P(u)+ alpha*(t x n).

Let's discuss it anyway in due time (the sooner v0.18 gets release, the sooner the v0.19 development cycle starts)...

---

### Comment by Kunda1 2019-09-12 21:28
Discussion thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=39269

---

### Comment by Kunda1 2020-04-22 11:12
Other discussion threads:
"[Sketch offset tool](https://forum.freecadweb.org/viewtopic.php?f=3&t=21983)"
"[Feature requests: Enabling FreeCAD for sewing pattern making](https://forum.freecadweb.org/viewtopic.php?f=8&t=45550)"

alitarraf, 2023-02-06:
any updates on this feature?

abdullahtahiriyo, 2023-02-07:
It was not implemented yet.

There has been one effort that is pending review as part of a very big PR that has many other features. I am unsure of how well it works as I have not tested it yet.

I cannot give you a lead time ATM. Sorry.

raindropsfromsky, 2023-08-07:
This feature is already added in a FreeCAD fork 18 months ago.
The developer "[PaddleStroke](https://github.com/PaddleStroke)" desires to merge his code with the upstream FreeCAD.

He has provided a [demo](https://www.youtube.com/watch?v=944AleKzzq0) of this tool.

https://github.com/PaddleStroke/FreeCAD/tree/ConstraintContextual

Can this be expedited?

Thanks in advance!

luzpaz, 2023-09-09:
CC @PaddleStroke

abdullahtahiriyo, 2023-10-03:
ETA is before the end of 2023. Hopefully in October/November.

PR #11174 (2023-11-03): [Sketcher] Tool Parameter widget + Offset tool
Fixes #10436
Fixes #7221
Fixes #5839

This PR adds parameters to sketcher tools. Some settings on the 3d view directly. Some settings in a taskbox widget on the taskview.

This PR include : 
- Tool Parameter framework.
- Point tool
- Line tool
- Rectangle tool
- Offset tool
- Polygon tool