FreeCAD/FreeCAD #5843: Milling simulation continuously stealing window focus and releases it

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3714

* **Reporter:** sleeply4cat
* **Date submitted:** 12/5/2018
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** assigned
* **Tags:** #post-to-forum, Path

# Original report text

When Path simulation is running, FreeCAD continuously takes window focus for a short time and then releases it. It makes the whole system unresponsive. 

# Additional information

Checked on Ubuntu 18 Cinnamon/Ubuntu 18 Gnome.
xev log:

```FocusOut event, serial 37, synthetic NO, window 0x3e00001,
    mode NotifyGrab, detail NotifyNonlinear

FocusIn event, serial 37, synthetic NO, window 0x3e00001,
    mode NotifyUngrab, detail NotifyNonlinear

KeymapNotify event, serial 37, synthetic NO, window 0x0,
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   

FocusOut event, serial 37, synthetic NO, window 0x3e00001,
    mode NotifyGrab, detail NotifyNonlinear

FocusIn event, serial 37, synthetic NO, window 0x3e00001,
    mode NotifyUngrab, detail NotifyNonlinear

KeymapNotify event, serial 37, synthetic NO, window 0x0,
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
```

FreeCAD about:
```OS: Ubuntu 18.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15302 (Git)
Build type: Release
Branch: master
Hash: 2e03d2f298677b8212c22cbbc3cb20b7c80eabb5
Python version: 2.7.15rc1
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)```

# Steps to reproduce

1. Open FreeCAD
2. Create a sample object
3. Select Path workspace
4. Create a job with some pathes
5. Start a simulation
6. Open another window (Gedit, for example)
7. type '1234567890', something like '156780' will appear

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Ubuntu 18.04.1**
* **Platform:** Linux
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2018-12-05 13:23
heads up sliptonic mlampert

---

### Comment by sliptonic 2019-01-23 00:38
shaiseger Do you have anything to add?

---

### Comment by Kunda1 2020-01-08 18:18
shaiseger ping

luzpaz, 2022-03-21:
@sliptonic is this something using camiotics will solve?

sliptonic, 2022-03-21:
It will be one more tool in the simulation and visualization toolbox.  Camotics simulation is much more mature than the FreeCAD native version but it is 3 Axis only for now.

luzpaz, 2022-03-21:
This reminds me of the TestWB issue. If you run a general test the focus continually gets pulled back to the tests and it's not possible to get much done (unless you're on dual monitors and can mouse click quickly :rofl: )

sliptonic, 2022-06-30:
They're pretty different things.   I'm planning to do a video on the Camotics thing.

xtemp09, 2023-10-26:
I've failed to reproduce the problem described.

The MainWindow becomes a bit unresponsive, though.

<details><summary>Ubuntu 22.04.3 LTS</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34784 (Git)
Build type: Debug
Branch: main
Hash: 61c1b2bfc2339b272d3fa44bb9dbb6bdba528a64
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

</details>

xtemp09, 2023-10-27:
@luzpaz, can you reproduce the problem?

luzpaz, 2023-10-27:
Sure, can you provide a quick test file ?

luzpaz, 2023-10-27:
@shaise can you confirm this is fixed ?

sliptonic, 2024-05-07:
Shaise is working on a complete replacement of the simulator.  Closing this issue in anticipation of that.

closed 2024-05-07 by sliptonic

luzpaz, 2024-05-07:
Thanks sliptonic. Can you link to the work for posterity purposes ?

shaise, 2024-05-08:
1. https://github.com/FreeCAD/FreeCAD/pull/13884
2. https://forum.freecad.org/viewtopic.php?t=87174