FreeCAD/FreeCAD #5853: Open file dialog freezes the laptop for several minutes (approx. 2 min) <-- Due to missing Code Signing

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3744

* **Reporter:** sbisbip
* **Date submitted:** 12/21/2018
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #ready-for-migration, macOS

# Original report text

Each time I use Freecad app the open file dialog freezes the machine for at least 2 min...
Please note at the beginning the open file is small like a rendering problem and after the frozen delay it has a normal scale.
If I browse the filesystem I have the same kind of freezes :(

# Additional information

my freedcad
OS: Mac OS X
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13541 (Git)
Build type: Release
Branch: releases/FreeCAD-0-17
Hash: 9948ee4f1570df9216862a79705afb367b2c6ffb
Python version: 2.7.14
Qt version: 5.10.1
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

# Steps to reproduce

How to reproduce:
1. Launch freecad app (wait at least 30 sec before the app is opened)
2. open file dialog
=> the open file dialog is small
=> the machine is not usable only the mouse cursor is visible, a keyboard action has no effects
3. wait about 2 minutes
=> the app is available
3.b browse the file system
=> same behavior as step 3

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **OS: macOS Mojave 10.14.2**
* **Platform:** macbook pro
* **Updated:** 2/3/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2018-12-24 12:28
Currently FC's MacOS maintainers haven't been active. Please post this issues to the forums to generate further discussion. 
Also can you reproduce this on [0.18_pre](https://github.com/FreeCAD/FreeCAD/releases/tag/0.18_pre)?

---

### Comment by sbisbip 2018-12-25 11:37
Hi Kunda1,

Thanks a lot for your message.
I have tested the dev version as you advised in your previous message.

With this following version the delay is shorter about 5 seconds, it is not friendly but it is acceptable.
I propose to decrease the severity.

Tested on:
OS: macOS 10.14
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15310 (Git)
Build type: Release
Branch: (HEAD detached at c88a64e8e)
Hash: c88a64e8e7a3bb66a23b965b655b5a5040dfa4d2
Python version: 2.7.15
Qt version: 5.11.2
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

---

### Comment by martijnthe 2019-01-31 10:07
I think it has something to do with the binary release not being code signed.
If I use codesign to sign the FreeCAD.app and all the executables in it, the freezing goes away.

Perhaps it's a good idea to sign the binary releases for macOS (by the CI build machine).

See my forum post here: https://forum.freecadweb.org/viewtopic.php?f=4&t=31343&p=283600#p283600

---

### Comment by Kunda1 2019-02-10 15:44
CC blacey peterl94

---

### Comment by michaelhund 2019-11-12 20:09
I can report this issue continues with FreeCAD-0.19-18644 running on MacOS 10.14.6 on a MacBook Pro mid 2014 model. The delay affects all file operations, and last about 45 seconds.

---

### Comment by Kunda1 2019-12-09 14:00
michaelhund thanks for the note. Please continue discussion on the forum https://forum.freecadweb.org/viewtopic.php?f=4&t=31343&p=283600#p283600 to find a solution with the community

luzpaz, 2022-06-21:
Related to https://github.com/FreeCAD/FPA/issues/6

PapsOu, 2022-10-18:
I also have this problem on a Linux fedora. When opening the file dialog, the UI is frozen few seconds and the dialog is a black square (not rendered).

```
fedora 35 : 5.19.15-101.fc35.x86_64

OS: Fedora Linux 35 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.1.Unknown
Build type: Unknown
Python 3.10.6, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.3, OCC 7.5.0
Locale: French/France (fr_FR)
Installed mods: 
  * Design456 0.0.1 (Disabled)
```

[image]

PapsOu, 2022-10-18:
And it seems that it is not related only to FreeCAD 0.20, because it happen with version 0.19 too...

michaelhq, 2023-01-20:
I had the same issue on my Windows 11 22H2 machine. 

I applied this workaround: https://github.com/realthunder/FreeCAD_assembly3/issues/371

By setting both parameter to false, it uses now the systems file dialog and not the Qt file dialog. No lagging or freezing anymore.

I think it is a Qt issue.

PapsOu, 2023-01-20:
https://bbs.archlinux.org/viewtopic.php?pid=2061934#p2061934

It seems that deleting the config file `~/.config/QtProject.conf` solves the problem. So not really related to Freecad, but related to Qt instead

michaelhq, 2023-01-20:
On Windows, deleting the Registry Key HKEY_CURRENT_USER\Software\QtProject\ solves the problem for the file open and export dialog, but not for the font selecting dialog. 

So for all Windows users, I recommend setting both parameters in "Tools -> Edit Praremeters" under "Preferences -> Dialog" to false until the issue with Qt is solved.

adrianinsaval, 2023-07-25:
is this still a problem for anyone?

michaelhq, 2023-07-25:
On Windows, without the workaround, yes it is still a problem.

luzpaz, 2023-08-06:
Should probably be added to the v1.0 milestone

chennes, 2023-08-06:
This ticket is not a single issue: it appears that there is a Windows issue that will be solved wholly independently of the original Mac bug report. I suggest closing this one (the Mac problem is solved by signing the code), and then opening an OS-specific ticket regarding the Qt problem and marking it "upstream" (presumably solved by our migration to Qt6).

luzpaz, 2023-08-07:
> opening an OS-specific ticket regarding the Qt problem and marking it "upstream" (presumably solved by our migration to Qt6).

Qt is cross-platform, why are we opening a specific ticket for each OS then? Unless I misunderstood and you meant to say 'OS-independent ticket' ?

chennes, 2023-08-07:
As far as I can tell from the above discussion, there is a problem on Windows that does not appear on other platforms, right?

luzpaz, 2023-08-07:
Closing in favor of #10099

closed 2023-08-07 by luzpaz