FreeCAD/FreeCAD #5854: Solver failing on certain sketch

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3745

* **Reporter:** chrisb
* **Date submitted:** 12/23/2018
* **FreeCAD version:** 0.18
* **Category:** General
* **Status:** assigned
* **Tags:** 

# Original report text

[Forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=32967)

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** General
* **Updated:** 2/6/2021

luzpaz, 2022-04-25:
Per user1234 on the forum:
> As far as i know, abdullah worked on the solver over this time. Now It works, also the redundancy is noticed by the solver. Should be solved. 
[image]

CC @abdullahtahiriyo @PLChris

PLChris, 2022-04-25:
Fixed.

closed 2022-04-25 by PLChris

luzpaz, 2022-04-25:
@PLChris any idea in what commit ?