FreeCAD/FreeCAD #5862: Add Crowdin API Issue lister to Crowdin scripts

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3777

* **Reporter:** Kunda1
* **Date submitted:** 1/17/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** translation

# Original report text

Using the [Crowdin API](https://support.crowdin.com/api/issues/) a Crowdin admin can view all open issues that are created through the Crowdin translation interface. This can help to solve user experience issues etc...

# Other bug information

* **Priority:** low
* **Severity:** feature
* **Category:** Feature
* **Updated:** 10/26/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2021-03-01 18:25
hyarion would you be interested in this task aswell. Right now it's a real PITA to traverse the crowdin UI to see open issues. This feature could be a game changer.

comradekingu, 2022-02-11:
The real question is why those aren't open to browse to begin with…

luzpaz, 2022-02-11:
@comradekingu They should be for logged in users on crowdin (via the interface)

luzpaz, 2022-04-03:
Crowdin -> Discord webhooks: https://github.com/SwitchbladeBot/crowdin-discord-webhooks

luzpaz, 2023-10-24:
@kaktusus anyone you think may be interested in helping with this ?

kaktusus, 2023-10-25:
The issues can be viewed by anyone logged in regardless of their privileges.

https://crowdin.com/project/freecad/activity-stream

[image]

[image]

Each authorization level has an individual WEB interface also for management. I don't know all of them.
_The presented solution is universal._
And has anyone looked into the manual?

kaktusus, 2023-10-25:
When I switch to another type of WEB editor (as an equalizer) I have a different set of tools available.

[image]

Of course _(I assume this is the case)_ this filter should also be available to the translator. Can someone with translator privileges confirm this?

[image]

here I can get a view for issues that have not been solved for selected languages (_max 10 at a time)_.

kaktusus, 2023-10-25:
I posted your question in the Translators forum: https://forum.freecad.org/viewtopic.php?style=4&t=82181