FreeCAD/FreeCAD #5864: Feature request sketcher - overload of the horizontal/vertical distance constraints between curves

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3780

* **Reporter:** freecad-heini-1
* **Date submitted:** 1/18/2019
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

https://forum.freecadweb.org/viewtopic.php?p=280421#p280421
https://forum.freecadweb.org/viewtopic.php?p=280680#p280680

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

adrianinsaval, 2023-03-28:
@FlachyJoe what do you think of this request? If you consider it feasible and potentially useful it might be worth keeping, otherwise I think this one should be closed.

for convenience I'll upload here the demo gif from the form post:
[image]

FlachyJoe, 2023-03-29:
I add this to my todo list.

NXTler, 2023-08-26:
This would make the sketcher so much better

luzpaz, 2023-09-10:
@FlachyJoe do you have a branch associated with this ticket yet ?

maxwxyz, 2024-01-05:
@FlachyJoe do you already have something, else I might try this one at some time.

maxwxyz, 2024-01-05:
To be clear, should the minimum or the maximum of the distance be constrained?

FlachyJoe, 2024-01-06:
@maxwxyz I have nothing yet, feel free.

maxwxyz, 2024-01-06:
@FlachyJoe ok, do you have a maximum distance between two circles already? Currently  the shortest distance between two circles is constrained, but I guess with this FR, the maximum distance between two curves is more interesting

FlachyJoe, 2024-01-06:
@maxwxyz you have to define a new constraint for the solver,  read Abdullah's [FreeCAD Sketcher Solver Architecture](https://github.com/FreeCAD/FreeCAD/files/13462593/fc-solver-latex.pdf) and take a look at https://github.com/FreeCAD/FreeCAD/blob/37f7ccfc862ebaae72769dbd75079be2b69bdaaf/src/Mod/Sketcher/App/planegcs/Constraints.cpp#L2923-L2977

maxwxyz, 2024-01-09:
@FlachyJoe so am I right to define 15 new constraints for circles and arcs:
- max distance circle to circle, circle to line, and circle to point
- max/min horizontal distance circle to circle, circle to line, and circle to point
- max/min horizontal distance circle to circle and circle to line

closed 2024-02-21 by maxwxyz