FreeCAD/FreeCAD #5865: impossible to remove several references from FEM references lists

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3782

* **Reporter:** uwestoehr
* **Date submitted:** 1/20/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** assigned
* **Tags:** FEM

# Original report text

- create a FEM shell plate thickness and add several reference surfaces
- now you have e.g. 10 references in the list and need to remove 8 of them. Thus you try to select them

result: You cannot select more than one reference in the list, neither by pressing the Ctrl or Shift key or by mouse dragging etc. So you have to right-click on every reference to delete it -> this is very annoying and time consuming

Attached is a screenshot as example. I need to delete there Face 11 to Face 17.

This issue appears for all FEM reference lists (e.g. also for the FEM material).

# Other bug information

* **Priority:** normal
* **Severity:** tweak
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by uwestoehr 2019-01-20 03:43
It would at least be helpful if pressing DEL would remove the references.

---

### Comment by Kunda1 2019-01-22 23:36
uwestoehr please follow the guidelines (enormous yellow banner at the top of the page) when opening tickets. There is no extensive FC version info, no forum thread. Please open a thread to discuss this with bernd, the FEM maintainer 

---

### Comment by uwestoehr 2019-01-23 23:31
> There is no extensive FC version info

Kunda1, I specified that I use version 0.18, see this bug report header. I also use the latest development snapshot as given in the yellow banner when reporting bugs.

> no forum thread. Please open a thread to discuss this with bernd, the FEM maintainer.

Sorry, but I don't have that much time. I reported a bug I saw and checked that it is not yet reported as best as possible. feel free to ignore my bug reports or close them immediately if you don't think it is a real bug.

I am an Open Source software developer too and know that some bug reports are not valid, thus I checked carefully not to report already existing issues. But you cannot expect people to discuss at first in a forum - waiting for replies, respond, waiting again... So I collected all bugs I found and reported them that they are not forgotten. (I don't use FC regularly, maybe once a month.)

---

### Comment by Kunda1 2019-01-25 17:22
My response ~0012509

Deferring to berndhahnebach

---

### Comment by berndhahnebach 2019-01-26 21:18
Hi uwe and welcome to FreeCAD FEM. I must admit you are right. All the reference shape gui work flow needs some love. It is on my TODO for a long time. But not in very front position ... 

cheers bernd

---

### Comment by berndhahnebach 2019-01-26 21:28
related #3783

---

### Comment by uwestoehr 2019-01-27 04:02
Hi Bernd, thanks for having a look. I also found other FC issues and tried to go the way to open a forum discussion first. But whenever I want to post something, I only get a message
"You cannot make another post so soon after your last."
What Am I doing wrong? Why can't I post anything to the forum despite I successfully created and activated an account?

---

### Comment by uwestoehr 2019-01-27 04:03
Hi Bernd, thanks for having a look. I also found other FC issues and tried to go the way to open a forum discussion first. But whenever I want to post something, I only get a message
"You cannot make another post so soon after your last."
What Am I doing wrong? Why can't I post anything to the forum despite I successfully created and activated an account?

---

### Comment by wmayer 2019-01-27 13:39
> What Am I doing wrong? Why can't I post anything to the forum despite I successfully created and activated an account? 

After posting the very first time on the forum a moderator must approve your initial post. This is because there are a lot of spammers these days and this way a spammer cannot flood the forum with crap.

---

### Comment by uwestoehr 2019-01-27 15:02
> After posting the very first time on the forum a moderator must approve your initial post.
OK, but please inform new people about this. I did not get any information despite the senseless message that I cannot post soon again despite not a single post was yet made.
I propose a message like this after pressing the first time Submit:
"Thanks for your first post. To avoid spam the first post of new users will be reviewed by our maintainers. Until it was approved you cannot made further posts."
Should I report this issue as a separate bug or post this in the forum?

---

### Comment by uwestoehr 2020-10-24 21:57
Meanwhile the situation improved. On can now use DEL to delete references. However, selecting several references and deleting them at once does not yet work. I will have a look.

donovaly, 2022-02-07:
Fixed with #5117

luzpaz, 2022-04-13:
@donovaly can we close this ticket? (And others that are 'FIxed for 0.20') ?

donovaly, 2022-04-14:
My intention was to keep them open until the fix is released. This way affected users will find the issue and not open a new one. When they find it and see it is already fixed, they could try out our weekly build.

However, there is no rule so far on closing issues.

donovaly, 2022-04-14:
But OK, FC 0.20 is near, so let's close the 0.20 fixes.

closed 2022-04-14 by donovaly