FreeCAD/FreeCAD #5866: several faces cannot be added at once in some FEM dialogs

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3783

* **Reporter:** uwestoehr
* **Date submitted:** 1/20/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** assigned
* **Tags:** FEM

# Original report text

- add a FEM solid material
- You need to select e.g. 20 faces for this material, therefore you open the material dialog

result: there is no way to add more than one face at once. So you can click on Add and then click on a face while keeping Ctrl pressed to select another face because the first face you select is always directly added. Therefore it is extremely annoying if you have to add tens of faces.

This bug does not occur in other FEM dialogs, like e.g. the constraint displacement dialog. There you can select all faces and finally press Add to get them all added at once.

The bug occurs in the following FEM dialogs:
- ElementGeometry2D
- SolidMaterial
- FemConstraintForce
- FemConstraintForc

The bug does not occur in these FEM dialogs:
- FemConstraintFixed
- FemConstraintDisplacement
- FemConstraintPressure
- FemConstraintTemperature
- FemConstraintHeatflux

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-01-22 23:36
uwestoehr please follow the guidelines (enormous yellow banner at the top of the page) when opening tickets. There is no extensive FC version info, no forum thread. Please open a thread to discuss this with bernd, the FEM maintainer. OR else what happens is that these tickets create tracker-bloat. Thanks for your comprehension.

---

### Comment by uwestoehr 2019-01-23 23:31
> There is no extensive FC version info

Kunda1, I specified that I use version 0.18, see this bug report header. I also use the latest development snapshot as given in the yellow banner when reporting bugs.

> no forum thread. Please open a thread to discuss this with bernd, the FEM maintainer.

Sorry, but I don't have that much time. I reported a bug I saw and checked that it is not yet reported as best as possible. feel free to ignore my bug reports or close them immediately if you don't think it is a real bug.

I am an Open Source software developer too and know that some bug reports are not valid, thus I checked carefully not to report already existing issues. But you cannot expect people to discuss at first in a forum - waiting for replies, respond, waiting again... So I collected all bugs I found and reported them that they are not forgotten. (I don't use FC regularly, maybe once a month.)

---

### Comment by Kunda1 2019-01-25 17:21
My response ~0012509

Deferring to berndhahnebach

---

### Comment by berndhahnebach 2019-01-26 21:27
~0012525 the mentioned love includes to make all reference shape gui work the same way in all constraints ...

---

### Comment by uwestoehr 2020-10-24 22:00
Still the case. I'll have a look.

donovaly, 2022-02-07:
Fixed by https://github.com/FreeCAD/FreeCAD/pull/5117

donovaly, 2022-06-24:
FC 0.20 was released -> fixed