FreeCAD/FreeCAD #5869: Volumetric milling simulation of arcs creating artifacts

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3790

* **Reporter:** sleeply4cat
* **Date submitted:** 1/24/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** assigned
* **Tags:** #post-to-forum, Path

# Original report text

The volumetric simulation of gear-like objects is pretty inaccurate:
!(https://cdn1.savepice.ru/uploads/2019/1/24/5adbdefccfa8270e974e5823cfe5c218-full.png)
Looks like it caused by a defect of the arc's end cup: its angular size two times smaller than it should be.

# Additional information

OS: Ubuntu 18.04.1 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15694 (Git)
Build type: Release
Branch: master
Hash: 96dee91d5e7d7fd37f43a4dcf406a120ae1f1b8a
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

These changes can correct the situation, but I have paws..

```--- a/src/Mod/Path/PathSimulator/App/VolSim.cpp
+++ b/src/Mod/Path/PathSimulator/App/VolSim.cpp
@@ -638,7 +638,7 @@ void cStock::ApplyCircularTool(Point3D & p1, Point3D & p2, Point3D & cent, cSimT
 	{
 		Point3D cupCirc(xynorm.x * r, xynorm.y * r, 0);
 		float rotang = (float)SIM_WALK_RES / r;
-		int ndivs = (int)(3.1415926535 / rotang) + 1;
+		int ndivs = (int)(2 * 3.1415926535 / rotang) + 1;
 		if (!isCCW)
 			rotang = -rotang;
 		cupCirc.SetRotationAngleRad(rotang);```

# Steps to reproduce

1. create path contains a gear (or any non-smooth path with "left" turn like [this](https://cdn1.savepice.ru/uploads/2019/1/24/64097f330a5fbd1052bd5170505f85a6-full.png)), test project is attached
2. start a simulation

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Ubuntu 16.04, 18.04**
* **Platform:** Linux
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-01-25 17:09
deferring to mlampert

---

### Comment by mlampert 2019-01-25 23:25
I'm not familiar with the simulation framework in path. I believe shaise would be the best person to look at this.

---

### Comment by Kunda1 2019-01-25 23:26
Assigning to shaiseger

---

### Comment by Kunda1 2019-01-25 23:29
sleeply4cat can you do us a favor and please post first to the forum to discuss? The bugtracker doesn't get as much traffic as the forum. This is obvious from the lack of activity on #3714

---

### Comment by Kunda1 2020-01-08 18:19
shaiseger ping

sliptonic, 2022-03-21:
The image links from mantis are broken.  Unless OP can repost images or the proposed code looks useful to @shaise,  I propose closing this issue.

luzpaz, 2022-03-21:
CC @sleeply4cat sorry about our bugtracker losing the original example file. Can you reproduce this issue at all?

sleeply4cat, 2022-03-23:
@luzpaz
Sorry, still can't access PC with original screenshots.
The issue is pretty reproducible at 0.19: https://www.youtube.com/watch?v=Go12zv2aQtY
Also, there is a similar issue with 3d pockets caused by this bug: where mill goes forward and down, it should cut the material by the bottom-rear side which isn't simulated.
What about that old patch.. Idk lol, it's pretty thoughtless fix, but it was working for me.

luzpaz, 2023-02-13:
@shaise do you mind weighing in real fast ?

sliptonic, 2023-08-12:
closing for lack of engagment.  Please reopen if there's anything left here.

closed 2023-08-12 by sliptonic