FreeCAD/FreeCAD #5870: Solver fails to find redundancy

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3791

* **Reporter:** chrisb
* **Date submitted:** 1/24/2019
* **FreeCAD version:** 0.18
* **Category:** General
* **Status:** assigned
* **Tags:** 

# Original report text

[Forum discussion.](https://forum.freecadweb.org/viewtopic.php?f=3&t=33679)

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** General
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by bejant 2019-09-11 15:48
Here is another example of the Solver (in this example, DogLeg) failing to find redundancy.
https://forum.freecadweb.org/viewtopic.php?f=3&t=39233#p333307
File is attached to the first post of the forum topic.
Steps to reproduce:
Open Sketch and drag a vertical line segment a bit.
Apply a Horizontal Distance Constraint of 8 mm to the horizontal line segment.
Apply Tangency Constraints between the arc and the vertical line segments.
Delete whichever Vertical Constraint the Solver flags as redundant.
Result: 1 DOF remains (the horizontal lone segment can be moved vertically) but the Solver fails to detect that the Horizontal Distance Constraint of 8 mm is redundant.

maxwxyz, 2024-12-07:
@AjinkyaDahale is this still relevant?

FreeWilly2022, 2025-03-21:
This appears to have been resolved, tested in 1.1.0dev build 40504 aarch64. Copying a sketch to another body results in "Link(s) to objects(s) ... go out of allowed scope" because the sketch being copied referenced other sketches in the original body. This maybe? resulted in the features in the original body being lost. It did not result in lost features when I did these steps.

The correct action would be to make a new body, select the sketch to be copied, and then paste the sketch and its referenced sketches to the new body. I tried this and it worked.

closed 2025-03-21 by maxwxyz