FreeCAD/FreeCAD #5871: Input of floating point values with point ('.') instead of comma (',')

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3795

* **Reporter:** markus51
* **Date submitted:** 1/25/2019
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** feedback
* **Tags:** units

# Original report text

I use FreeCad 0.17 with Win7, in Germany. 
The country settings uses a comma (',') for floating point values.

Many peoples usees a point to input floating point valuses,
but FreeCad needs sometimes a comma, or sometimes are comma and points allowed.

Combo-View/Model:                only comma 
Combo-View/Tracks/Constaints:    only comma 
Spreadsheet:                     comma and points

I would be nice when always both floating formats will be accepted.

 

# Steps to reproduce

1.) Create a new sketch and add some constraints with a length
2.) Double click in Combo-View/Tracks/Constraints on the new item
3.) Input "12.3" and press ENTER
4.) The new value is "123" instead of "12.3"

11.) Create a new sketch and add some constraints with a length
12.) Double click in Combo-View/Model on a Constraint
13.) Input "12.3" and press ENTER
14.) The new value is "123" instead of "12.3"

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **OS: Windows **
* **Updated:** 11/23/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-01-25 23:16
markus51 Have you tested first in 0.18dev before posting (as we ask in the enormous yellow banner at the top of the page)?
We also ask very clearly at the top of the page not to post bugs (especially v0.17) unless they are verified in the forum.
Please download 0.18dev and verify #3794 #3795 #3796

---

### Comment by Kunda1 2019-01-25 23:16
markus51 Have you tested first in 0.18dev before posting (as we ask in the enormous yellow banner at the top of the page)?
We also ask very clearly at the top of the page not to post bugs (especially v0.17) unless they are verified in the forum.
Please download 0.18dev and verify #3794 #3795 #3796

---

### Comment by markus51 2019-01-28 12:38
The same problem exists on version 0.18dev

---

### Comment by uwestoehr 2019-01-28 13:04
As a new user I stumbled over the same problem.
I get this with:

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15711 (Git)
Build type: Release
Branch: master
Hash: 5caae5f430237a73cd17e283d45b3ac9ab005feb
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: German/Germany (de_DE)

---

### Comment by Kunda1 2019-01-29 18:17
CC wmayer

---

### Comment by wmayer 2019-02-21 16:48
When using a standard QDoubleSpinBox from the Qt library then it also changes the number to 123 when you enter 12.3.
This is because a dot in German locale is the group separator and not a decimal separator and the dot is only a visual aid for the user. When the user leaves the widget then it automatically removes the dot.
The QuantitySpinBox is a FreeCAD widget and follows the same logic as Qt's QDoubleSpinBox. So, there is no need to change anything with the current behaviour.

---

### Comment by markus51 2019-02-21 19:01
I have fixed this ugly stuf in my private version (only 30 min work)
Now point and comma will be accepted.

That is user-friendliness !

---

### Comment by uwestoehr 2019-02-22 00:20
> Now point and comma will be accepted.

Very good. However I learned meanwhile that some German users use the dot to separate thousands. E.g. they input "18.000" and don't want to have it treated as "18,000" but as "18000". Therefore I understand wmayer that FC should not add a workaround but let Qt do the job.

---

### Comment by markus51 2019-02-23 12:30
German bakers usualy use the comma and sometimes a point to separate thousands.

German engineers use in the most cases the point. The geraman technical software which is writen in my company also uses the point for floating point values. Other technical software from other companies usualy to the same.

If we talk about financial software, you're right. In technical software, it looks a little different.

---

### Comment by uwestoehr 2019-02-24 00:23
> If we talk about financial software, you're right. In technical software, it looks a little different.

I recently realized that my engineer colleagues use µm as base unit instead of mm because we build microsystems. Always entering values like "0,060" annoyed them. So now they can write simply "60" and if there is a mm length, they then write 12.000.

Many engineering software offers to use the decimal separator of the UI language. That is why I did not yet noticed that - I use in most cases the English UI. Yesterday I made a test and setting the UI to German I have to use the comma. So things are changing in the software development and I think it makes sense to consistently follow the rules of the language.
The same is with FC. Use the English locale and you can use the dot as decimal separator.

Therefore I vote to mark this issue as already fixed since you can use FC with English UI andg get the dot as you like.

---

### Comment by wmayer 2019-02-24 17:48
[quote=markus51]That is user-friendliness ![/quote]
It's user-friendly for you personally but this does not mean that's user-friendly for everybody and it's at the cost of consistency.

Then in FreeCAD in dozens of places Qt's QDoubleSpinBox is used which strictly follows the locale settings of the OS. How do you deal with this, then?

---

### Comment by uwestoehr 2019-02-24 18:15
wmayer I notice that there is an inconsistency because in tables FC automatically changes inputs like "2,3" to "2.3". Should I open a new tracker issue for that?

---

### Comment by uwestoehr 2019-03-01 15:02
> Should I open a new tracker issue for that? 

I did it: #3875

Since the spreadsheet issue is now covered I think this bug report could be closed.

---

### Comment by Kunda1 2019-03-12 02:30
Closing ticket

---

### Comment by uwestoehr 2020-03-24 13:23
The discussion about this popped up again:
https://forum.freecadweb.org/viewtopic.php?f=3&t=35927
thus reopen the ticket for now.

---

### Comment by uwestoehr 2021-11-02 14:57
Was there any decision made for this?

openBrain, do you maybe know something since you are assigned to a related bug?

---

### Comment by openBrain 2021-11-02 17:35
uwestoehr : yes, the PR that was pending has been integrated and I required for testing here : [https://forum.freecadweb.org/viewtopic.php?f=8&t=62546](https://forum.freecadweb.org/viewtopic.php?f=8&t=62546)
AFAIK there is no regression declared ATM, but the fix may still be a bit incomplete. From the feedback, it works well for QSpinBox-like widgets, but not for example for Gui::InputField type.
I have to investigate to be sure this latter type can only hold "formulas" but no "expressions" before I extend the fix.

---

### Comment by uwestoehr 2021-11-02 18:30
Thanks for your work!

> it works well for QSpinBox-like widgets

Not for me, I reported this now in the forum thread to be discussed further.

---

### Comment by openBrain 2021-11-16 17:54
uwestoehr : new version has been integrated in daily that should work well in all situations. You can test again and report if needed. ;)

---

### Comment by uwestoehr 2021-11-23 21:38
How can I test this? I added the Boolean "SubstituteDecimalSeparator" under General in the parameters, then restarted FC but cannot see any difference. So e.g. an input of "3,0" results in 3 mm, but "3.0" results in 30 mm.

(By the way, reading my own statements from years ago, it is interesting how I changed my mind.)

0penBrain, 2022-02-07:
Has been solved by #5150. Ticket was awaiting @donovaly feedback to be closed.

donovaly, 2022-02-07:
The last info I have is that this issue is not resolved:
- take me as a German, my Windows is a German one, therefore it uses the comma as decimal separator.
- I use FreeCAD in English and therefore want the dot as decimal separator

As it is I have no setting to achieve this.

0penBrain, 2022-02-07:
@donovaly : okay, this is a wider problem than just inputing. If you look at the field values in your case, it uses German format also to display.
Your issue could be summarized as "FreeCAD uses number format of OS regional settings, not one of choosen language".
I'd be in favor of closing this issue and opening a new one describing specifically your issue. ;)

donovaly, 2022-02-07:
OK, then please open a new bug and I close this one now.

closed 2022-02-07 by donovaly