FreeCAD/FreeCAD #5872: Shortcuts "V1" .. "V7" to change the "Draw-Style" didn't work in sketch view

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3796

* **Reporter:** markus51
* **Date submitted:** 1/25/2019
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** confirmed
* **Tags:** shortcuts

# Original report text

The shortcuts "V1" .. "V7" to change the "Draw-Style" didn't work in the sketch view.

When i change to "Part Design" the shortcuts "V1" .. "V7" working correctly.

# Steps to reproduce

1.) Open a sketch.
2.) Press "V"+"1" --> the "Draw-Style" din't change
3.) Press "V"+"4" --> the "Draw-Style" din't change
4.) Change in the toolbar to "Part Design" 
5.) Press "V"+"1" --> the "Draw-Style" changes correctly
6.) Press "V"+"4" --> the "Draw-Style" changes correctly

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Bug
* **OS: Win7 **
* **Updated:** 12/16/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-01-25 23:22
markus51 Have you tested first in 0.18dev before posting (as we ask in the enormous yellow banner at the top of the page)?
We also ask very clearly at the top of the page not to post bugs (especially v0.17) unless they are verified in the forum.
Please download 0.18dev and verify #3794 #3795 #3796

---

### Comment by markus51 2019-01-28 12:00
The same problem exists on version 0.18dev

---

### Comment by Kunda1 2019-01-29 18:19
Making an 0.18 target. Devs please review (and decide if it should be in the 0.18 or 0.19 roadmap) Thanks. 

---

### Comment by yorik 2019-02-20 13:18
This is because the shortuct key "V" is used by the vertical constraint. So V1, V2... never happens.

I don't see any obvious solution here, either remap V1-->V7 to something else or remap V to something else...

---

### Comment by Dargazo 2019-02-22 23:10
Because the "V" shortcut is a subset of the "V + n" (with n = 1, 2, 3, ...) shortcut, I am thinking that we might need to have a look at the [activatedAmbiguously](https://doc.qt.io/archives/qt-4.8/qshortcut.html#activatedAmbiguously) QT signal. Maybe this will give more insight. Perhaps it is triggered, but no SLOT is attached?

I'd like to test this out myself, but after having tried to compile + debug run the current FreeCAD 0.18 version multiple times today (Windows 10 + VS 2013 + libpack), I am done for today :)

---

### Comment by chennes 2021-12-16 21:33
These no longer map to the vertical constraints, but are now being used for the view orientation. This is still not the "draw style", however.

qume, 2023-01-31:
Not sure if this is the right place to ask, but a button to toggle between wireframe and shaded in sketcher would be really handy

Roy-043, 2023-07-15:
In the version listed below the V1 etc. shortcuts work while editing a sketch. No need to switch to the PartDesign WB. Therefore closing this.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33492 (Git)
Build type: Release
Branch: master
Hash: 6e235a4809e14674aa2cd23e68630b68998a53d4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

closed 2023-07-15 by Roy-043