FreeCAD/FreeCAD #5876: aggregative function in the Schedule  value field

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3802

* **Reporter:** danreb
* **Date submitted:** 1/27/2019
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

any chance aggregative functions could be used inside schedule-way of computing stuff ?

e.g. in the value field being able to use this kind of expression:
max(edge.Length for edge in object.Shape.Edges)
?

cf

# Additional information

linked forum thread
https://forum.freecadweb.org/viewtopic.php?f=8&t=31190&p=259272#p259272

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2023-02-27:
@yorikvanhavre also should we transfer to BIM workbench ?

yorikvanhavre, 2023-02-28:
@luzpaz No, it's (currently :) ) pure Arch.

Thinking better about this issue, I have no clear idea on how to implement that ATM... Maybe pass through some other expression engine? Input welcome ;)