FreeCAD/FreeCAD #5877: Improvement of "Constraints" list in the Sketch task of the "Combo view"

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3803

* **Reporter:** markus51
* **Date submitted:** 1/27/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** feedback
* **Tags:** #post-to-forum, UI

# Original report text

Some features to improve the "Constraints" list.

1.) Add a "Hide all constraints" to the "Constraints" listbox
2.) Add a "Show all constraints" to the "Constraints" listbox
3.) Add a shortcut Cltr+A to select all item in the "Constraints" listbox
4.) Add a shortcut Cltr+A to select all item in the "Elements" listbox

# Steps to reproduce

1.) Create a new sketch
2.) Draw some elemnts
3.) Goto the "Combo view" (in "Sketcher")
4.) Open the task tab

# Other bug information

* **Priority:** low
* **Severity:** tweak
* **Category:** Feature
* **OS: Win7 **
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2019-01-29 18:30
Where is the forum thread? 
JFYI, tickets like this will get lost if they aren't discussed in the forum. The community decides if a ticket should be opened. Thank you for your comprehension.

---

### Comment by Kunda1 2019-04-06 19:05
abdullah Can I move this to 'Sketcher' ?

---

### Comment by abdullah 2019-04-12 17:16
kunda1

Sure. It is sketcher. I have just moved it.

---

### Comment by abdullah 2019-06-22 05:58
kunda1
markus51

There are a couple of aspects that would benefit from community discussion:

1) Selection of all elements via language dependent shortcut (e.g. ctrl+A)

normal list shortcuts are disabled, as the lists are intended to interact with the keyboard commands of the sketcher.

The checkbox in the constraints is specific to visibility, not selection. This has to be taken into account.

However, it may be possible to add a button to select all items. Maybe a context menu entry. Here I think UI experts, which I am not, could have interesting ideas/alternatives.

2)If by "constraints listbox" the context menu of the constraints is intended, it is very easy to add. Again, maybe UI experts come with a better usability solution.

It would not be the first time I propose a UI and people come with great enriching ideas. I have learnt to leverage the knowledge of the community in this cases. I think it is important it gets discussed.

adrianinsaval, 2023-03-28:
Most of this is implemented, in the constraint list the Ctrl+A shortcut works but not in the elements widget, could you comment on that @PaddleStroke @abdullahtahiriyo ?

abdullahtahiriyo, 2023-05-01:
The elements widget is a complicated non-standard list. Selection is quite special. 

We could default to selecting only edges. And first point if GeomPoint. It is quite low in my priority list. If somebody wants to take a look how to implement this they are welcome.

I am even considering merging the elements and constraints widgets during the next development cycle. With layers it is possible to hide a single geometry. There are observations that constraints should not be visible if the geometry is not visible (the contrary is not necessarily true). If it finally happens it will come after revamping the elements widget.