FreeCAD/FreeCAD #5880: Docker container for FreeCAD development

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3821

* **Reporter:** Kunda1
* **Date submitted:** 2/8/2019
* **FreeCAD version:** 
* **Category:** General
* **Status:** assigned
* **Tags:** 

# Original report text

https://forum.freecadweb.org/viewtopic.php?f=4&t=34046&p=285576#p285576

[quote=looo]
I tried to make something similar once, but this wasn't successful https://github.com/FreeCAD/FreeCAD_Conda/blob/master/utils/fc_conda_docker/cached_build/Dockerfile. It would be nice to have a docker container including all the dependencies of freecad, an ide (like kdevelop), git, and the freecad source. Ideally everything should be compiled already (maybe in debug-mode).

So to get started with freecad development all one has to do is:

`docker run --name freecad-dev freecad/freecad-development`

and this starts the ide.
This would be soooo useful.
[/quote]

# Other bug information

* **Priority:** low
* **Severity:** feature
* **Category:** General
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by David_D 2019-02-12 17:50
I'm working on this [here](https://gitlab.com/daviddaish/freecad_docker_env), and working on a docker container with the up-to-date dependencies [here](https://gitlab.com/daviddaish/freecad_docker_env/tree/updates).

---

### Comment by Kunda1 2019-02-12 20:00
David_D fantastic. I've elevated your privileges since this is a very important task. Cheers!

---

### Comment by David_D 2019-03-04 04:22
I have gotten the docker image to a level that I am satisfied with it as a proof of concept. Kunda1, would you, or some other interested party be good enough to test the docker image on their machine?

I am aware it is missing the WebView and FEM modules. I will add them later.

Instructions of it's use is available on it's [readme](https://gitlab.com/daviddaish/freecad_docker_env).

---

### Comment by Kunda1 2020-02-01 14:48
Update: https://forum.freecadweb.org/viewtopic.php?f=4&t=42954

adrianinsaval, 2023-05-04:
there are dockerfiles available at the main repo for quite some time now

closed 2023-05-04 by adrianinsaval

luzpaz, 2023-05-05:
> there are dockerfiles available at the main repo for quite some time now

Please link to them

adrianinsaval, 2023-05-05:
https://github.com/FreeCAD/FreeCAD/tree/master/tools/build/Docker