FreeCAD/FreeCAD #5882: Support of AXIS2_PLACEMENT_3D entities in STEP files

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3834

* **Reporter:** wmayer
* **Date submitted:** 2/13/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** STEP

# Original report text

A STEP file can contain AXIS1_PLACEMENT_3D or AXIS2_PLACEMENT_3D entities that should be read-in when loading such a file.

# Additional information

https://forum.freecadweb.org/viewtopic.php?f=3&t=34162

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by wmayer 2019-02-24 21:04
See also: http://opencascade.wikidot.com/entitynames

---

### Comment by wmayer 2019-02-24 21:53
The named AXIS2_PLACEMENT_3D are part of a PRESENTATION_LAYER_ASSIGNMENT entity.

---

### Comment by Kunda1 2019-10-04 00:52
I've assigned this ticket to realthunder per https://forum.freecadweb.org/viewtopic.php?f=3&t=34162&p=338358#p338358

dpasukhi, 2024-09-30:
That functionality already planned and named as a "Supplemental geometry".
Already implemented on 7.5 custom branch.
See: https://tracker.dev.opencascade.org/view.php?id=33203

luzpaz, 2024-09-30:
@dpasukhi 
[image]
Clicking 'Proceed' just leads to general bugtracker dashboard, 'my_view_page.php'

dpasukhi, 2024-09-30:
Updated.
And a new issue was connected: https://tracker.dev.opencascade.org/view.php?id=33206

luzpaz, 2024-10-15:
@dpasukhi any reason why I can't view both links logged in anonymously to the occt bugtracker ?

dpasukhi, 2024-10-15:
My apologies. I changed project, but forgot visibility.
Now it should work.

luzpaz, 2024-10-16:
@dpasukhi Thank you! Can you please explain what is the difference between 7.5 custom branch and the current 7.8.1 stable release ?

dpasukhi, 2024-10-16:
The customer's functionality was quick implications (1+-day). Not final implementation.
I plan to rework that, but all the time forget :(
Now I remember, but don't have resources, I will try to include into 7.9

leoheck, 2025-02-26:
Opencascade 7.9.0 was released. 
https://dev.opencascade.org/content/open-cascade-technology-790-released

dpasukhi, 2025-02-26:
Supplemental geometry were out of scope for 7.8.0.
Axises are possible to read startibg from 7.6.0 but only as a part of PMI reference and it will be included into model, not separate from geometry.
Supplemental=constructive geometry still only on client branch.
I will allocate interns to migrate rgar functionality.
Bit it can be only on the one of the next releases

stefano2734, 2026-09-16:
See https://github.com/Open-Cascade-SAS/OCCT/pulls?q=is%3Apr+state%3Aclosed+supplemental+geometry

Occt 7.9.3 available Or test with occt 8.0.1.