FreeCAD/FreeCAD #5884: Cursor info in Sketcher doesn't use proper units

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3849

* **Reporter:** sopwith
* **Date submitted:** 2/18/2019
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** #tobeclosed, units

# Original report text

When drawing a shape in Sketcher, e.g. a line or a rectangle, the units displayed in blue next to the cursor are always mm, even if I've selected another units system in Preferences -> General -> Units.

# Additional information

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15860 (Git)
Build type: Release
Branch: master
Hash: 3b708c7f84b0425076b520e1d95627b20fd75fe0
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

# Steps to reproduce

Hopefully the above is self-explanatory.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Windows 10**
* **Platform:** x64
* **Updated:** 2/6/2022

# Discussion from Mantis ticket

### Comment by wmayer 2019-02-19 12:58
Check ViewProviderSketch::setPositionText
Check all occurrences of setPositionText in CommandCreateGeo.cpp

---

### Comment by Kunda1 2022-02-04 19:20
openBrain we can close this, right?

---

### Comment by openBrain 2022-02-06 17:45
Kunda1 : Nope. It's still a valid bug.

Syres916, 2023-07-12:
Just tested using:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33575 (Git)
Build type: Release
Branch: master
Hash: 2559d7e7b63081116a2b3509c2230bcbef8ebf88
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.3.0
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.3
  * A2plus
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * MakerWorkbench 1.0.1
  * freecad.gears 1.0.0
  * Curves 0.6.8
  * toSketch 1.0.1
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.56
  * sheetmetal 0.2.63
```

Using different unit schemas the cursor text changes as expected, @luzpaz can you please confirm and close/update as required.

grubertechnic, 2023-07-14:
@Syres916 : a short test works for me too. This commit dd14b42829a1ca2c3cccd9f417d028dacd26664c should be the one, that includes that functionality (last year). I candate to close.

closed 2023-07-15 by luzpaz