FreeCAD/FreeCAD #5889: FEM, vtk result pipeline, drag and drop for filter objects

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3879

* **Reporter:** berndhahnebach
* **Date submitted:** 3/3/2019
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

it would be helpful if filter obj would support drag and drop in and out of pipeline objects.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by berndhahnebach 2019-03-03 20:21
python example code for calculix cantilever example.
- load example
- delete pipeline object
- create a new pipeline object
- use this code to move the warp inside the pipeline

`App.ActiveDocument.Pipeline.Filter = [App.ActiveDocument.WarpVector]`

---

### Comment by berndhahnebach 2019-03-03 20:22
C++ drag and drop example for analysis container

https://github.com/FreeCAD/FreeCAD/blob/2b0d3477cb5700f5188be0ee5152e50ad6766e9c/src/Mod/Fem/Gui/ViewProviderAnalysis.cpp#L188-L211

---

### Comment by berndhahnebach 2019-03-03 20:23
FreeCAD 0.18.16028

ickby, 2025-05-18:
This issue is solved since the pipeline multiframe update, drag and drop should work as expected.

luzpaz, 2025-05-18:
@ickby do you mind referencing the PR that fixed this ? (if you know it?)

ickby, 2025-05-18:
It was PR #19610. 

I double checked:
1. It does support dragging pipelines with their filter in and out of a analysis
2. It supports dragging filters withing a pipeline for reordering, and also into/out of branches
3. It supports dragging filters between pipelines
4. It does NOT support dragging filters out of pipelines in the analysis. I did not implement that as it seems  not to make sense for me why this should be useful or needed.

I think if draging out of pipelines was intended as "deactivate" we should rather make the filters suppressible, that fits better to the FEM workbench logic and is way cleaner.

closed 2025-05-18 by luzpaz