FreeCAD/FreeCAD #5891: FEM, add group info to the GUI tool show mesh information

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3883

* **Reporter:** berndhahnebach
* **Date submitted:** 3/5/2019
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

see summary

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2023-11-26:
@FEA-eng would you mind improving this ticket ? @berndhahnebach didn't elaborate much here.

FEA-eng, 2023-11-26:
> @FEA-eng would you mind improving this ticket ? @berndhahnebach didn't elaborate much here.

I don't know what he meant here but it's clearly about this tool: https://wiki.freecad.org/FEM_MeshDisplayInfo

Perhaps he was thinking about including there the information from the results of this tool (that creates groups of mesh entities but seems to be broken or poorly documented: #10441): https://wiki.freecad.org/FEM_MeshGroup

So if there were mesh groups defined, the mesh info tool would show them (probably their statistics like the number of nodes/elements).

FEA-eng, 2025-02-18:
> Right now, the group are generated in the internal logic, but do not appear in any export file on my machine, as I have checked multiple times using gmsh.

Mesh groups work properly but get exported only to selected file formats as mentioned in the documentation: https://wiki.freecad.org/FEM_MeshGroup#Notes

TheFaither, 2025-02-18:
> Mesh groups work properly but get exported only to selected file formats as mentioned in the documentation: https://wiki.freecad.org/FEM_MeshGroup#Notes

Then I must be failing completely to catch them. I have tried using both mentioned formats, and nothing seems to Appear in gmsh. Can I ask you how have you validated the export? I’d like to learn it myself and I’ll be happy to expand the wiki page if I learn anything

FEA-eng, 2025-02-18:
> Then I must be failing completely to catch them. I have tried using both mentioned formats, and nothing seems to Appear in gmsh.

Did you regenerate (reapply) the Gmsh mesh after defining groups ? Then it should work. If you open such a file in text editor or import it to software supporting such meshes, you should see the groups.

TheFaither, 2025-02-18:
[image]

[image]

`Prepare meshing...
Start meshing...
Info    : Running 'C:\Program Files\FreeCAD 1.0\bin\gmsh.exe -v 4 - C:\Users\**\AppData\Local\Temp\fcfem_mtcnptsv\shape2mesh.geo' [Gmsh 4.13.1, 1 node, max. 1 thread]
Info    : Started on Tue Feb 18 13:23:48 2025
Info    : Reading 'C:\Users\**\AppData\Local\Temp\fcfem_mtcnptsv\shape2mesh.geo'...
Info    : Reading 'C:\Users\**\AppData\Local\Temp\fcfem_mtcnptsv\Connect_Geometry.brep'...
Info    : Done reading 'C:\Users\**\AppData\Local\Temp\fcfem_mtcnptsv\Connect_Geometry.brep'
Info    : Meshing 1D...
Info    : [  0%] Meshing curve 1 (Circle)
Info    : [ 20%] Meshing curve 2 (Line)
Info    : [ 30%] Meshing curve 3 (Circle)
Info    : [ 40%] Meshing curve 4 (Line)
Info    : [ 60%] Meshing curve 5 (Circle)
Info    : [ 70%] Meshing curve 6 (Circle)
Info    : [ 80%] Meshing curve 7 (Circle)
Info    : [ 90%] Meshing curve 8 (Circle)
Info    : Done meshing 1D (Wall 0.0052344s, CPU 0s)
Info    : Meshing 2D...
Info    : [  0%] Meshing surface 1 (Cylinder, MeshAdapt)
Info    : [ 30%] Meshing surface 2 (Plane, Delaunay)
Info    : [ 60%] Meshing surface 3 (Plane, Delaunay)
Info    : [ 80%] Meshing surface 4 (Cylinder, MeshAdapt)
Info    : Done meshing 2D (Wall 0.0465126s, CPU 0.015625s)
Info    : Meshing 3D...
Info    : 3D Meshing 1 volume with 1 connected component
Info    : Tetrahedrizing 330 nodes...
Info    : Done tetrahedrizing 338 nodes (Wall 0.0036258s, CPU 0s)
Info    : Reconstructing mesh...
Info    :  - Creating surface mesh
Info    :  - Identifying boundary edges
Info    :  - Recovering boundary
Info    : Done reconstructing mesh (Wall 0.006945s, CPU 0s)
Info    : Found volume 1
Info    : It. 0 - 0 nodes created - worst tet radius 2.43857 (nodes removed 0 0)
Info    : 3D refinement terminated (408 nodes total):
Info    :  - 0 Delaunay cavities modified for star shapeness
Info    :  - 0 nodes could not be inserted
Info    :  - 1394 tetrahedra created in 0.0042468 sec. (328247 tets/s)
Info    : 0 node relocations
Info    : Done meshing 3D (Wall 0.0206314s, CPU 0.015625s)
Info    : Optimizing mesh...
Info    : Optimizing volume 1
Info    : Optimization starts (volume = 227.033) with worst = 0.0729422 / average = 0.734753:
Info    : 0.00 < quality < 0.10 :         1 elements
Info    : 0.10 < quality < 0.20 :        24 elements
Info    : 0.20 < quality < 0.30 :        28 elements
Info    : 0.30 < quality < 0.40 :        25 elements
Info    : 0.40 < quality < 0.50 :        35 elements
Info    : 0.50 < quality < 0.60 :        97 elements
Info    : 0.60 < quality < 0.70 :       249 elements
Info    : 0.70 < quality < 0.80 :       370 elements
Info    : 0.80 < quality < 0.90 :       407 elements
Info    : 0.90 < quality < 1.00 :       158 elements
Info    : 53 edge swaps, 0 node relocations (volume = 227.033): worst = 0.23006 / average = 0.75241 (Wall 0.0006922s, CPU 0s)
Info    : 54 edge swaps, 0 node relocations (volume = 227.033): worst = 0.300136 / average = 0.752796 (Wall 0.0007561s, CPU 0s)
Info    : No ill-shaped tets in the mesh :-)
Info    : 0.00 < quality < 0.10 :         0 elements
Info    : 0.10 < quality < 0.20 :         0 elements
Info    : 0.20 < quality < 0.30 :         0 elements
Info    : 0.30 < quality < 0.40 :        24 elements
Info    : 0.40 < quality < 0.50 :        38 elements
Info    : 0.50 < quality < 0.60 :       106 elements
Info    : 0.60 < quality < 0.70 :       255 elements
Info    : 0.70 < quality < 0.80 :       373 elements
Info    : 0.80 < quality < 0.90 :       400 elements
Info    : 0.90 < quality < 1.00 :       157 elements
Info    : Done optimizing mesh (Wall 0.0016152s, CPU 0s)
Info    : Meshing order 2 (curvilinear on)...
Info    : [  0%] Meshing curve 1 order 2
Info    : [ 10%] Meshing curve 2 order 2
Info    : [ 20%] Meshing curve 3 order 2
Info    : [ 30%] Meshing curve 4 order 2
Info    : [ 40%] Meshing curve 5 order 2
Info    : [ 40%] Meshing curve 6 order 2
Info    : [ 50%] Meshing curve 7 order 2
Info    : [ 60%] Meshing curve 8 order 2
Info    : [ 70%] Meshing surface 1 order 2
Info    : [ 70%] Meshing surface 2 order 2
Info    : [ 80%] Meshing surface 3 order 2
Info    : [ 90%] Meshing surface 4 order 2
Info    : [100%] Meshing volume 1 order 2
Info    : Surface mesh: worst distortion = 0.907497 (0 elements in ]0, 0.2]); worst gamma = 0.439816
Info    : Volume mesh: worst distortion = 0.77704 (0 elements in ]0, 0.2])
Info    : Done meshing order 2 (Wall 0.0079329s, CPU 0.015625s)
Info    : 2496 nodes 2075 elements
Info    : Removing duplicate mesh nodes...
Info    : Found 0 duplicate nodes 
Info    : No duplicate nodes found
Info    : Writing 'C:\Users\**\AppData\Local\Temp\fcfem_mtcnptsv\Connect_Mesh.unv'...
Info    : Done writing 'C:\Users\**\AppData\Local\Temp\fcfem_mtcnptsv\Connect_Mesh.unv'
Info    : Done reading 'C:\Users\**\AppData\Local\Temp\fcfem_mtcnptsv\shape2mesh.geo'
Info    : Stopped on Tue Feb 18 13:23:48 2025 (From start: Wall 0.129741s, CPU 0.0625s)
Process finished`

I have added here some screenshots, the gmsh output on reapply, and then the .zip of the mesh. If you can see anything that I have done wrong, please let me know, and thank you for your help

[TwoFibresFused-FEMMeshGmsh.zip](https://github.com/user-attachments/files/18845107/TwoFibresFused-FEMMeshGmsh.zip)

FEA-eng, 2025-02-18:
@TheFaither Can you share the .FCStd file with this model?

TheFaither, 2025-02-18:
Well, now I feel stupid! Of course, how did I not think about it...

[TwoFibresFused.zip](https://github.com/user-attachments/files/18845301/TwoFibresFused.zip)

FEA-eng, 2025-02-18:
> Well, now I feel stupid! Of course, how did I not think about it...
> 
> [TwoFibresFused.zip](https://github.com/user-attachments/files/18845301/TwoFibresFused.zip)

When I export the mesh to .unv from this file, I have the group there.

[image]

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

But it would be better if you could create a new issue or write on the forum to continue the discussion so that we don't go too off-topic here.

TheFaither, 2025-02-18:
Thank you, finally I understood how to check if it worked! I think this should be added to the wiki. Now my problem is understanding why gmsh is not seeing the group, since it can open and work with the .unv file no problem, but that is out of the scope of FreeCAD. 

I highly suggest adding to the wiki that the feature can be checked by visualising the file via text editor, web searches were not helping about it and gmsh or other mesh visualisers do not seem to show anything about this grouping mechanism.

FEA-eng, 2025-02-19:
Ok, I added this information to the wiki.