FreeCAD/FreeCAD #5894: Counterbore and countersink with top clearance

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3893

* **Reporter:** chrisb
* **Date submitted:** 3/8/2019
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

According to this [forum discussion](https://forum.freecadweb.org/viewtopic.php?f=8&t=34663) it is sensible to extend thePartDesign hole feature:
- Countersink should have an additional (counterbore) depth at the top
- Countersink and Counterdepth should both provide the possibility for top clearance (reversed through all).

This image is linked from the forum discussion:
!(https://forum.freecadweb.org/download/file.php?id=78214)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by eivindkvedalen 2019-04-11 18:18
Branch for test: https://github.com/eivindkv/FreeCAD/tree/Fix_issue_3893

---

### Comment by reox 2019-04-13 12:26
Additional features:

* Set the depth of the (useable) thread (Usually this can be made in combination with the hole depth, see below)
* Option: Set the depth of the bore without the angled point, i.e. set the useable bore depth instead of the total depth.

I know there are some rules for tapped blind holes how deep the core hole should be and how long the threaded part ist. But I can not find them right now...
I think it was like useable thread + 4 times the screw pitch or something like that.
The reason is that a normal tap has some non cutting threads in the beginning.
Other CAD packages:
http://ww3.cad.de/foren/ubb/upl/B/BenediktKuepper/BohrungsundGewindetiefe.JPG
http://www.ask-hochreiter.de/Info/PM12/D/Gewinde00.gif

The Counterbore + Countersink parts could be also available for tapped holes - in German this is called "Stufenbohrung"
Like this:
https://ww3.cad.de/foren/ubb/uploads/Eduard+Beser/esfunzt.jpg
(Although the question is if this is really needed...)

Another useful feature would be to have two default counterbore settings: A shallow one (which is the current counterbore setting) and a deeper one, where the head of a ISO4672 screw is completly hidden.

luzpaz, 2022-03-17:
The code from @eivindkv is at https://github.com/eivindkv/FreeCAD/tree/Fix_issue_3893
specifically this commit: https://github.com/eivindkv/FreeCAD/commit/a3fe08aa0cb7fe6ea91908a97f246d90b734949c

luzpaz, 2022-03-18:
@donovaly i know we're in a feature freeze, but if you have a moment, would you take a look at this and see if you think it should be considered for v0.21 ?

donovaly, 2022-03-18:
@eivindkv , can you please make an PR in our FreeCAD repository (against current master)?

antonok-edm, 2024-03-31:
I believe this can be closed as it's already addressed by https://github.com/FreeCAD/FreeCAD/pull/7562?

benj5378, 2025-03-15:
@luzpaz @maxwxyz Could you check if this is already fixes