FreeCAD/FreeCAD #5895: Translation: "displacement" in ship.ts and fem.ts have different meaning and need differentiation in the sourcecode

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3902

* **Reporter:** kazan417
* **Date submitted:** 3/12/2019
* **FreeCAD version:** 0.18
* **Category:** General
* **Status:** new
* **Tags:** FEM, ship, translation

# Original report text

I found some bad translation in fem.ts. First of all displacement (водоизмещение) automatically copied by crowdin from ship.ts, but here should be different word - смещение. This is true for many languages, not just Russian.
I think we can solve it by changing "displacement" in ship.ts source string to "ship displacement"

# Other bug information

* **Priority:** normal
* **Severity:** text
* **Category:** General
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by kazan417 2019-03-12 17:12
https://forum.freecadweb.org/viewtopic.php?f=10&t=32573&p=291900#p291900

bopoh13, 2022-05-27:
It was indicated #2845 that transfers would not be applied through PR. _The link in the second message doesn't work._
https://github.com/FreeCAD/FreeCAD/blob/ee6f0d0ba9dcb2301f09229753d3e87b0d4d8530/src/Mod/Part/Gui/Resources/translations/Part_ru.ts#L4044-L4045
```diff
-<translation type="unfinished">Сначала выберите фигуру для революции.</translation>
+<translation>Сначала выберите фигуру для вращения.</translation>
```
https://github.com/FreeCAD/FreeCAD/blob/0cf9a778a54cf4aff76863ec1bf4c265b2596ef0/src/Mod/PartDesign/Gui/Resources/translations/PartDesign_ru.ts#L4528-L4529
```diff
-<translation type="unfinished">&amp;Sketch</translation> 
+<translation>&amp;Эскиз</translation> 
```
https://github.com/FreeCAD/FreeCAD/blob/0cf9a778a54cf4aff76863ec1bf4c265b2596ef0/src/Mod/PartDesign/Gui/Resources/translations/PartDesign_ru.ts#L4553-L4554
```diff
-<translation type="unfinished">Apply a pattern</translation>
+<translation>Применить шаблон</translation>
```
Hotkey hints may also not work.

luzpaz, 2022-05-27:
@sanguinariojoe can you weigh in on this? I know it's not your native language, but nevertheless, applies to Ship WB