FreeCAD/FreeCAD #5898: missing feedback when exporting to FEM mesh formats

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3906

* **Reporter:** uwestoehr
* **Date submitted:** 3/14/2019
* **FreeCAD version:** 0.18
* **Category:** Bug
* **Status:** assigned
* **Tags:** export, FEM, VTK

# Original report text

- open an analysis
- select the "CalculiX_static_results" in the model tree
- export it to VTK

result: this is error-prone:
Users might accidentally not choose "FEM result VTK" but "FEM mesh formats".
Yes, a result is no mesh, however in the export dialog these 2 entries are above each other so one can quickly make a mistake. The problem is that the user don't get any info. He exports but nothing will in fact be exported. No file will be written and he doesn't get info that nothing was output.
If possible, if the user selected a result object either the export to mesh option won't be available or he gets the info that the selection is not compatible with the selected object.

# Additional information

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=18&t=34815&start=10#p293375

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16078 (Git)
Build type: Release
Branch: master
Hash: 67df9f3a828fbb0fec29d039792d4baf3d9ee8e0
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

FEA-eng, 2024-02-18:
@maxwxyz Can you remove the Bug label? This one is a feature request.

FEA-eng, 2024-02-27:
@maxwxyz Actually, it seems that this one can be closed. There is an error message when doing what's described in the issue:

[image]

Maybe it was added at some point without closing the issue. Anyway, the behavior is as requested now.

closed 2024-02-27 by maxwxyz